seemoo-lab/internalblue #1: Bluetooth Debug Build for LineageOS 14.1

max-weller, 2018-09-09:
I just compiled the bluetooth debug library for LineageOS 14.1, the steps are slightly different than for AOSP:

Follow the build setup steps according to https://wiki.lineageos.org/devices/hammerhead/build until the Start the build section. Then do:

    cd system/bt/
    bdroid_CFLAGS='-DBT_NET_DEBUG=TRUE' mma -j4

----

flex crashes on Ubuntu 18.04 - workaround:

    export LC_ALL=C

https://stackoverflow.com/questions/49301627/android-7-1-2-armv7

demantz, 2018-09-10:
Thank you! I added the instructions to the README.md file in android_bluetooth_stack (https://github.com/seemoo-lab/internalblue/commit/401f492dd4fbbb24d4763c5380d85a78ab76a619).

Also thanks for providing the compiled .so in the pull request! Much appreciated!

closed 2018-09-10 by demantz

abbiy, 2022-09-19:
i am getting error while i do brunch hammerhead
cdm : brunch watch hammerhead
09:17:44 - error: Initialization error - The directory does not seem to be a Brunch project. No package.json was found. That's where Brunch plugins need to be listed. Plugins tell Brunch how to compile your files, so without any plugins Brunch won't know what to do. To get started, run `npm init` to generate a package.json file and `npm install --save-dev javascript-brunch css-brunch` to install and save plugins for regular JS and CSS. The directory does not seem to be a Brunch project. No package.json was found. That's where Brunch plugins need to be listed. Plugins tell Brunch how to compile your files, so without any plugins Brunch won't know what to do. To get started, run `npm init` to generate a package.json file and `npm install --save-dev javascript-brunch css-brunch` to install and save plugins for regular JS and CSS.