seemoo-lab/internalblue #4: _sendThreadFunc: No response from the firmware.

wrlu, 2019-01-14:
Hi~
I have already watched your excellent 35C3 talk Dissecting Broadcom Bluetooth and try to do the similar task on my own Nexus 5. However I received some python error when I try to connect to my phone. I have already sync the AOSP with branch `android-5.1.1_r3` and recompile the bluetooth stack `bluetooth.default.so` with `bdroid_CFLAGS='-DBT_NET_DEBUG=TRUE'` parameter. The Nexus 5 can enable bluetooth in the system settings properly after I push the new dymanic library file to the path `/system/lib/hw/`. So where went wrong? 
Additional information:
OS version: Ubuntu 18.04 LTS amd64
Python version: 2.7.15rc1
Nexus 5 Android version: Android 5.1.1

And here is the traceback:
```sh
[*] Running hcitool with sudo...
[*] Found one HCI device, hci: B4:6D:83:86:70:21 (hci0)
 [?] Please specify device:
    => 1) adb: 07a6d474012a4eab (Nexus 5)
       2) hci: B4:6D:83:86:70:21 (hci0)
[*] Connected to 07a6d474012a4eab
[!] sendHciCommand: waiting for response timed out!
Traceback (most recent call last):
[!] _sendThreadFunc: No response from the firmware.
  File "/usr/local/bin/internalblue", line 11, in <module>
    load_entry_point('internalblue==0.1', 'console_scripts', 'internalblue')()
  File "/usr/local/lib/python2.7/dist-packages/internalblue-0.1-py2.7.egg/internalblue/cli.py", line 130, in internalblue_cli
    if not reference.connect():
  File "/usr/local/lib/python2.7/dist-packages/internalblue-0.1-py2.7.egg/internalblue/core.py", line 659, in connect
    self.initialize_fimware()
  File "/usr/local/lib/python2.7/dist-packages/internalblue-0.1-py2.7.egg/internalblue/core.py", line 679, in initialize_fimware
    if (u8(version[9]) != 0x00 or u8(version[8]) != 0x0f):
TypeError: 'NoneType' object has no attribute '__getitem__'
```

jiska2342, 2019-01-14:
Hi,

`[*] Connected to 07a6d474012a4eab` means that adb generally works.

In Android 6.0.1, the file that enables Bluetooth Snoop Log via TCP is in `system/bt/hci/src/btsnoop_net.c` with the listen port set to `8872`. Injecting Bluetooth HCI via TCP is enabled in `system/bt/hci/src/hci_inject.c` and set to the port `8873`. You can locally see these ports entering `netstat -t` on your Android phone.

It might be that this feature is not yet implemented in Android 5.1.1. Is there any reason why you need this specific build or would upgrading to Android 6.0.1 work for you?

prjpet, 2019-01-19:
Hi,

I have ran in to a similar issue except that I also get an additional error...:

```
 [!] sendH4: waiting for response timed out!
Traceback (most recent call last):
  File "/usr/local/bin/internalblue", line 11, in <module>
    load_entry_point('internalblue==0.1', 'console_scripts', 'internalblue')()
  File "/usr/local/lib/python2.7/dist-packages/internalblue-0.1-py2.7.egg/internalblue/cli.py", line 130, in internalblue_cli
    if not reference.connect():
  File "/usr/local/lib/python2.7/dist-packages/internalblue-0.1-py2.7.egg/internalblue/core.py", line 667, in connect
[!] _sendThreadFunc: No response from the firmware.
    self.initialize_fimware()
  File "/usr/local/lib/python2.7/dist-packages/internalblue-0.1-py2.7.egg/internalblue/core.py", line 687, in initialize_fimware
    if (u8(version[9]) != 0x00 or u8(version[8]) != 0x0f):
TypeError: 'NoneType' object has no attribute '__getitem__'

```
I used Lineage OS 14.1 but my Android version is 7.1.2. 

Could this be the issue?

jiska2342, 2019-01-19:
Hi @prjpet ,
the additional error is due to the update I pushed during the last days which enables H4 Broadcom messages, but it is basically the same issue.

Do I get you right that you used android_bluetooth_stack/nexus5_lineageos14.1/bluetooth.default.so on a Nexus 5 running Android 7.1.2? Have you already tried android_bluetooth_stack/android7_1_2/bluetooth.default.so which I just pushed some hours ago?

I'm currently rebuilding all the modules to enable the new H4 Broadcom messages, so just let me know your exakt build number listed on "Settings -> About phone".

prjpet, 2019-01-20:
Hi @jiska2342,

> Do I get you right that you used android_bluetooth_stack/nexus5_lineageos14.1/bluetooth.default.so on a Nexus 5 running Android 7.1.2? 
    - Yes. This combination produced the above error.

> Have you already tried android_bluetooth_stack/android7_1_2/bluetooth.default.so which I just pushed some hours ago?
    - Also yes, but the library wouldn't load properly. I received an error message when I tried to enable bluetooth in LineageOS: _Bluetooth shared stopped working._

I have now reverted to a backup of Android 6.0.1 (completely removed Lineage OS) and loaded the android_bluetooth_stack/android6_0_1/bluetooth.default.so from and everything seems to be working fine.

wrlu, 2019-01-20:
Thanks for your working~
I would like to try to run this project after I upgrade my Android version to 6.x or flash a new Lineage OS firmware. 
Waiting for your good news.

jiska2342, 2019-01-29:
@wrlu A few steps later in the process I realized that Android 5 only supports HCI sniffing but no HCI injection. So except from seeing a live HCI BTSnoop Log over network nothing changes :( Code difference in these parts between Android 5 and 6 is quite a lot. So Android 5.1.1 will be a won't fix until I find a lot of time to patch it. I just pushed the HCI sniffing only *bluetooth.default.so*.

closed 2019-01-29 by jiska2342