seemoo-lab/internalblue #6: Installed........but doesnt run.....

rootorben, 2019-07-04:
Hi.......im using LSW Kali.....
after installing deps and setup i tried to run the program......

but all i got was this

`dak47@ASUS:~$ sudo python2 -m internalblue.cli
[sudo] password for dak47:
   ____     __                    _____  __
  /  _/__  / /____ _______  ___ _/ / _ )/ /_ _____
 _/ // _ \/ __/ -_) __/ _ \/ _ `/ / _  / / // / -_)
/___/_//_/\__/\__/_/ /_//_/\_,_/_/____/_/\_,_/\__/

type <help> for usage information!

dpkg-query: no path found matching pattern *bin/armeabi*linux*-as*
[!] pwntools cannot find binutils for arm architecture. Disassembling will not work!
dpkg-query: no path found matching pattern *bin/armeabi*linux*-as*
[!] pwntools cannot find binutils for arm architecture. Disassembling will not work!
[ERROR] './adb' does not exist
[*] No adb devices found.
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/internalblue/cli.py", line 171, in <module>
    internalblue_cli()
  File "/usr/local/lib/python2.7/dist-packages/internalblue/cli.py", line 132, in internalblue_cli
    devices.extend(connection_method.device_list())
  File "/usr/local/lib/python2.7/dist-packages/internalblue/hcicore.py", line 115, in device_list
    for dev in self.getHciDeviceList():
  File "/usr/local/lib/python2.7/dist-packages/internalblue/hcicore.py", line 47, in getHciDeviceList
    s = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_RAW, socket.BTPROTO_HCI)
  File "/usr/lib/python2.7/socket.py", line 191, in __init__
    _sock = _realsocket(family, type, proto)
socket.error: [Errno 97] Address family not supported by protocol
dak47@ASUS:~$`

Some ideas?

jiska2342, 2019-07-04:
Still looks like a lot of broken dependencies, such as `binutils-arm-linux-gnueabi` and `adb`. 

*InternalBlue* can connect to a Broadcom Bluetooth chip as follows:

1. Patched Android/LineageOS smartphone with `adb`
1.  `AF_BLUETOOTH` socket on your local Linux device
1. Patched iOS smartphone (with additional command line switch)

*InternalBlue* fails to start because `adb` is not installed (no problem) and then continues with a local Bluetooth socket. Then, the actual problem appears:

`socket.error: [Errno 97] Address family not supported by protocol`

On Linux, there is a socket type `AF_BLUETOOTH`. I think this is defined somewhere on kernel driver level, and thus breaks when using the Windows Subsystem for Linux. I don't know if there is a proper fix for this. I would suggest to run *InternalBlue* in a virtual machine and pass your Bluetooth chip as USB device to the virtual machine.

derfalx, 2019-07-04:
With the WSL 2 system Microsoft is going to "integrate" a virtualized Linux Kernel into windows. Then the usage of AF_BLUETOOTH and others should be possible (see https //github.com/Microsoft/WSL/issues/2039#issuecomment-505201850).

rootorben, 2019-07-04:
❤

jiska2342, 2019-07-05:
I'm closing this issue since it is not a problem with *InternalBlue*, but with Bluetooth sockets on WSL. We might add Windows support in the future, though. Right now, a student makes good progress with integrating macOS. Adding more operating systems and chips is definitely one of our priorities.

closed 2019-07-05 by jiska2342