seemoo-lab/internalblue #7: Packet Injection question

Matheus-Garbelini, 2019-07-21:
Hello, I was wondering about the LMP packet injection feature. It's mentioned that in some firmwares it's possible to send invalid LMP. Is InternalBlue only limited to LMP or Bluetooth Low Energy Link Layer can also be injected with invalid raw packets?
Thanks

jiska2342, 2019-07-21:
Hi,

when we found the symbols for the firmware, we found that there is an HCI command provided by *Broadcom* that supports to inject LMP packets into an existing connection. This is what the `sendlmp` command supports right now.

Internally, this is implemented as follows by *Broadcom*: `bthci_cmd_vs_SendLmpPdu` calls `DHM_LMPTx`, which checks the frames prior to sending against validity. If you want to disable the validity check, you can execute `fuzzlmp` for the *Nexus 5* and *CYW20735* evaluation board. This is an Assembly patch and therefore not that easy to port to other devices.

As of now, we only support LMP. But if you need it and tell me which device you use for testing, I can also check if I can implement an LCP fuzzing routine for you. It should be relatively straight forward, the sending routine is `lmulp_sendLcp`/`dhmulp_LcpTx`. Similar to LMP, the LCP routine checks if packets are valid prior to sending them, which would be an additional patch, similar to `fuzzlmp`. Since there is no HCI command, I would need to add that, too.

Please tell me the specific device you are using and I will write a patch if I have the same platform. I would prefer Raspberry Pi 3+/4/3 or Cypress evaluation boards, if possible.

Matheus-Garbelini, 2019-07-22:
@jiska2342 I have a Raspberry Pi 3 and hanks very much for your fast response.

jiska2342, 2019-07-22:
Hi :) So, I just did first tests on a *Nexus 5* and it seems to work. Validity checks work a bit different in the LCP handler, you can pass various lengths for the same opcode by default, and everything turned out to be a bit easier than for LMP. I also pushed code for a *Raspberry Pi 3*, but I still need to test it tomorrow when I have access to one. If you want to test it already, use the `sendlcp` command, i.e., `sendlcp 0c090f00beef`.

Matheus-Garbelini, 2019-07-23:
@jiska2342 Thanks for the code, I'll do some tests with it and let you know.

jiska2342, 2019-07-23:
@Matheus-Garbelini I just pushed a new version, the one before wrote the patch to a memory area that is likely to also be used by other functions. It should work on Rasperry Pi 3/3+/4 by now.

For whatever reason, the current Raspian image does not identify the Broadcom chip properly or misses some information about it. Therefore, I was not able to debug the LCP traffic using `vendor_diag` and `btmon`. But at least, `sendlcp 0203`, which is `LL_TERMINATE_IND`, always terminated the connection.

Matheus-Garbelini, 2019-07-25:
@jiska2342 Thanks for the update. I've acquired a nordic NRF52 Sniffer so it's easy to see what's going on. The sendlcp command works nice with valid lcp commands. Highlighted in black is the custom lcp commands.
[image]

The only caveat is that it appear that some valid lcp commands are handled by the firmware in case of channel map update instance. Fortunately it's still possible to send malformed packets by sending less bytes. In this case it makes a connection loss then. So I'm not sure if this means that the firmware is miss-configuring the channel map or the other device is processing the invalid channel map and losing track of the connection instead:
[image]

In any case I'm closing this issue. I've read the InternalBlue paper and is surprising to me that injection of normal bluetooth/ble packets were not easily possible before. I though that firmware developers tested code with random packets also, but it seems it's not always the case :-). Once again, thanks for the new feature addition.

closed 2019-07-25 by Matheus-Garbelini

jiska2342, 2019-07-25:
Thank you so much for testing :) If you want to set the channel map and also apply it to the Raspberry Pi, try using HCI. There is a HCI command that works at least pretty well on the *CYW20735* evaluation board and is also specified in the standard:

`sendhcicmd 0x2014 0000ff0000`

Matheus-Garbelini, 2020-02-28:
@jiska2342 Thanks a lot for your insight on BLE technology. The situation is worse than I initially thought ([SweynTooth was disclosed on 10th February](https://asset-group.github.io/disclosures/sweyntooth/)), and I hope researchers all around the world can continue to secure wireless as a whole. Let's make sure to keep our Fitbits updated :thumbsup: .

Thanks and best regards.

jmr123456-beep, 2020-12-18:
@jiska2342 thanks a lot for you insight and effort.  However I have question. when I use CRY20735, how to communicate this with InternalBlue. I think There is no ADB in Evaluation Kit .

Matheus-Garbelini, 2020-12-18:
hi @jmr123456-beep , when using CRY20735, you need to use bluez as instructed on this internalblue guide:
https://github.com/seemoo-lab/internalblue/blob/master/doc/linux_bluez.md
Adb is just for patched android smartphones.

jurek4321, 2024-05-25:
> when we found the symbols for the firmware, we found that there is an HCI command provided by _Broadcom_ that supports to inject LMP packets into an existing connection. This is what the `sendlmp` command supports right now.

@jiska2342 Where I can find firmware with symbols?