seemoo-lab/internalblue #8: Monitoring LMP packets...

Aleberello, 2019-09-09:
Hi, first of all thank you for this interesting research.
I'm trying to use InternalBlue with my Sony Z3 Compact, which from what I understand it should have the same bluetooth chip as the Nexus 5.
I need this tool to experiment with the KNOB attack (https://github.com/francozappa/knob) but i'm having several problems, including the lack of the possibility to see LMP messages from Wireshark.

I have compiled and flashed on the device Android AOSP 6.0.1 following both your guide and the specific one from Sony.
I also compiled the bluetooth.default.so and flashed with the Debugging Features enabled.

At this moment InternalBlue seems to connect properly to the device (see the attached screens) but when I open the monitor i can't see any message on the LMP protocol layer.

[image]
[image]

I also attach the Wireshark capture.
[https://drive.google.com/open?id=1LJSv2HaEm92DLKmY7TtRAAe2VNDiGCtu](url)

What am I doing wrong?
Many thanks in advance,
Alessandro

jiska2342, 2019-09-09:
Hi @Aleberello,

looks like you combined different versions of things here.

Variant A: Update to our current mechanism
-------------------------------------------------------

Install a `bluetooth.default.so` with diagnostic logging enabled - this also requires a code change, not just recompiling, as documented. I think you did this already, since there is no warning displayed.

Then install our Wireshark dissector that can handle Broadcom diagnostic messages [from here](https://github.com/seemoo-lab/h4bcm_wireshark_dissector). LMP and LCP traffic will automatically be logged. No Assembly hooks required for this. I guess, you are simply missing this dissector.

I optimized the new KNOB testing tool for this, which I uploaded a few days ago in our [examples](https://github.com/seemoo-lab/internalblue/blob/master/examples/nexus5/KNOB_PoC.py) folder for different chip versions. It only sets one global variable and insures on addition that the key entropy will always be set to 1 in the according LMP message.

All the messages that appear when you do you `sendlmp` command are because you did not specify a connection handle. The `sendlmp` function will try to automatically determine a valid connection handle and pick the first active connection, or the default connection handle.

Variant B: Use the (outdated) original KNOB PoC
-------------------------------------------------------------
This approach will work no matter if it supports diagnostic logging or not. Daniele used our very first *InternalBlue* version for the KNOB PoC, which must be almost a year old by now. In this version, we did not know about the proprietary Broadcom diagnostic logging protocol. Instead, we installed our own Assembly patches that would follow the LMP packet logic and copy packets via HCI. LMP packets would be sent as vendor specific HCI event and be prepended with `LMP_` in ASCII. Porting this patch to other chips was very hard, that's why we removed it and moved to the more generic diagnostic logging, that works on various devices. As the packet format is different, you also need a different Wireshark dissector for this type of LMP logging.

So, in order to get Daniele's original code working, you also need to downgrade your *InternalBlue* installation and use a different Wireshark dissector. Unless you actually want to write Assembly patches for the LMP logic, I would not recommend this. In case you really want to write patches, you might consider the *Nexmon* `bluetooth-wip` branch.

Aleberello, 2019-09-09:
Thank you for answered me so fast.
So, i'm trying with the "Variant A" that you proposed. I've already installed the `bluetooth.default.so` with diagnostic logging enabled and your Wireshark dissector plugin.
The screens of my previous message were already referred to this set-up, now i've also tried to compile the  `bluetooth.default.so`  with the `android_receive_diagnostics.diff`, following your instructions:

> `git apply android_receive_diagnostics.diff`
> `bdroid_CFLAGS='-DBT_NET_DEBUG=TRUE' mma -j4`

At the moment, when i try to use any InternalBlue command (like `connect`), bluetooth crashes in the device causing consequently the crash of InternalBlue. 
Also Wireshark doesn't show me LMP packets anyway.

These are the screens of what described:
The first refer only to the command `connect`, it crashes a few seconds later after the connection with the other device.
[image]

This is what happens when i do quickly the command `sendlmp `after the connection.
[image]

I believe that this behavior is due to the changes in the file `android_receive_diagnostics.diff` (I took the one inside [this folder](https://github.com/seemoo-lab/internalblue/tree/master/android_bluetooth_stack/android6_0_1)) since with the compiled file without them it did not give me these crashes (although I was not able to see the LMP packages from Wireshark) as you can see from my first post.

I just can't understand if I'm doing something wrong or it's just a matter of having a device different from those supported.
Thanks for your patience,
Alessandro

jiska2342, 2019-09-09:
Hi @Aleberello,

from what you are doing it definitely looks correct. The device recognizes the firmware. Two suggestions for debugging:

1. Could you run *InternalBlue* with the previous `bluetooth.default.so` and give me the output of `hd 0x200400`? That should print the exact firmware version. Along with this, also `info patchram` could be interesting.

2. If the error is in `android_receive_diagnostics.diff` you should see some message parsing errors related to Bluetooth / H4 / serial / UART in `logcat` and `dmesg` on your smartphone.

Best,
Jiska

Aleberello, 2019-09-10:
Hi @jiska2342,

in [this temporary repo](https://github.com/Aleberello/tmp) i've uploaded the logs.
Specifically I've putted the commands of point **1** in "logs" and the logs of point **2** in "logs/android".
I hope it's the best solution to make it easier for you to read.

_whit_diff.txt_ and _whitout_diff.txt_ refer to the outputs of the commands executed using the compiled `bluetooth.default.so` with or without `android_receive_diagnostics.diff` (actually they are equal).

Have a nice day,
Alessandro

Aleberello, 2019-09-13:
Hi @jiska2342,
don't worry, you're already doing a lot to help me.

So, i've re-flashed `bluetooth.default.so` applying the `android_receive_diagnostics.diff`.

I've updated the [temporary repo](https://github.com/Aleberello/tmp/tree/master/logs) by inserting two logs and two screen videos containing: 
- the execution of the `connect` command only -> "android/connection_only_logcat" + "connect_only.mp4"
- the execution of the `connect` command plus the `sendlmp `command (executed before the crash) -> "android/connection_sendlmp_logcat" + "connect_sendlmp.mp4"

I don't know if I'm doing the log files well, to record them I'm simply using the command: 
`adb logcat  -v time > "log_path.log"`

Thank you,
Alessandro

jiska2342, 2019-09-16:
Hi @Aleberello,

as expected, the issue seems to be to assemble frames correctly including diagnostic messages. Your error looks a bit different from what I expected, but it definitely is of this category:

    --------- beginning of main
    09-13 16:09:22.808 W/bt_hci  ( 2454): transmit_downward legacy transmit of command. Use transmit_command instead.
    09-13 16:09:24.810 W/bt_hci  ( 2454): transmit_downward legacy transmit of command. Use transmit_command instead.
    09-13 16:09:26.584 W/bt_l2cap( 2454): L2CAP got conn_comp for unknown BD_ADDR
    09-13 16:09:26.613 W/bt_hci_packet_fragmenter( 2454): reassemble_and_dispatch got continuation for unknown packet. Dropping it.
    09-13 16:09:34.606 E/bt_hci  ( 2454): command_timed_out hci layer timeout waiting for response to a command. opcode: 0x41f
    09-13 16:09:34.607 E/bt_hci  ( 2454): command_timed_out restarting the bluetooth process.

This basically means that due to receiving the first LMP diagnostic message during connection establishment, the remaining HCI messages coming from the chip could not be reassembled. So, my `android_receive_diagnostics.diff` does not apply to the version of AOSP you checked out from the repositories. As the HCI message parser relies on getting four message types (`0x01` for HCI commands, `0x04` for HCI events, `0x02` and `0x03` for ACL/SCO, all with correct length information included), it fails when parsing diagnostic messages (starting with `0x07`, fixed length packets). The parser has no chance to sync to the next HCI message start again, as there is no preamble or similar. The whole Bluetooth daemon and chip need to be reset.

Depending on what you want to do with the KNOB test, you do not need to see LMP. It just works, and at least on iPhones you will see a warning for sure. But diagnostic messages would definitely be an advantage for debugging.

Aleberello, 2019-09-20:
Hi @jiska2342,
oh ok, correct me if I'm wrong, 
without the `android_receive_diagnostics.diff` the tool works but I can't see the transmitted packages LMP because the parser can't understand how to interpret the diagnostic packages (starting with `0x07`).
So if I just wanted to test that the device I want to associate with is vulnerable to KNOB, I can do it without using the diff file, right?

Three things are still not clear to me:
1.  The KNOB Attack Test that you implemented can I also use it for my device right?
2.  Is the patch inserted to handle entropy during negotiation only valid for the current internalblue session?
3. Do I have any chance of seeing, without the LMP packages, some information on the negotiation of the key?

I'm sorry but these are slightly technical things for me, I struggle to understand them completely. :)

Thank you SO much!

jiska2342, 2019-09-20:
Hi @Aleberello,

1. Yes, you can use it on your device. You have a firmware that the patch works for. (I also ported it to the Raspberry Pi 3 this week btw...)

2. The patch works until you reset Bluetooth, i.e., turn it off and on again, or reboot your device. Depending on your Android version, rebooting is a safer option than just turning Bluetooth of and on.

3. Yes! Via HCI.

From the original KNOB paper:

>  The application can interact with the firmware using the HCI protocol. In particular, it can use the HCI Read Encryption Key Size request, to check the amount of negotiated entropy *after* the Bluetooth connection is established and theoretically abort the connection. This check is neither
required nor recommended by the standard as part of the key negotiation protocol.

I think this is the approach being used on *iOS 12.4* which has a patch that works across all iPhones.

You can perform this test yourself as follows. The `HCI_Read_Encryption_Key_Size` command is defined in the Bluetooth 5.1 specification, page 1123. Its parameter is a 2 byte `Connection_Handle` and it returns a 1 byte `Status`, the same 2 byte `Connection_Handle` you passed and 1 byte `Key_Size`. If you have *InternalBlue* opened while you establish a connection, it tells you the `Connection_Handle`.

Open *InternalBlue* and start Wireshark.

    wireshark start

In Wireshark, add a filter `bthci_evt.opcode == 0x1408`, this will make things easier to see. Then, establish a connection, just as you would usually do using the interface on your smartphone, or by pairing from the other device to your smartphone. You should get a notification that the connection was established.

    [*] [Connection Create initiated]
    [*] [Connect Complete: Handle=0xc  Address=ab:cd:ef:00:11:22  status=Success]

Use that connection handle to ask for the key size, which is the key entropy. Works with one byte here, as it is little endian, but you might need to add two zeros. For me it worked like this:

    sendhcicmd 0x1408 0x0c

In Wireshark, when expanding the HCI event, you would see something like `Encryption Key Size: 16` if there was no KNOB attack.

Then you quit *InternalBlue*.

    quit

And apply the KNOB script:

    python examples/nexus5/KNOB_PoC.py 

You open *InternalBlue* again, connect again, send the same HCI command again. If I do this against an up-to-date *Samsung Galaxy S8* (last update 3 days ago) it happily continues with connection establishment at least for some moments without a warning. Then, Wireshark shows `Encryption Key Size: 1`. Ooops. ;)

Happy debugging and testing :)

Aleberello, 2019-09-27:
Here i'am @jiska2342,
sorry, but I've been a little busy with other issues these days.

I tried to execute your instructions step by step (very understandable, thank you) but I haven't managed to get anything out of it..

I have tried both with old devices and with updated devices but when I try to request the length of the key I get inconsistent results, for now it always returns me:
- `Encryption Key Size: 0 ` -> [/2f 0c 00 1c] 
or
- `Encryption Key Size: 28` -> [/2f 0c 00 00]

This happens after and before I apply `python examples/nexus5/KNOB_PoC.py`.

Now I really don't think they are correct values, it seems strange to me that it gives me 0 as the key length.

Is it possible that it is related to this thing you wrote to me?

> Use that connection handle to ask for the key size, which is the key entropy. Works with one byte here, as it is little endian, but you might need to add two zeros.

As always, thank you,
Alessandro

jiska2342, 2019-09-28:
Hi @Aleberello,

as this is probably less intuitive than I thought, I added a new version of the PoC with automatic callbacks ([same file](https://github.com/seemoo-lab/internalblue/blob/master/examples/nexus5/KNOB_PoC.py)). It works as follows:

```
    -----------------------KNOB-----------------------
    Automatically continuing on KNOB interface...
    Use the 'knob' command to *debug* the attack, i.e.:
        knob --hnd 0x0c
    ...shows the key size of handle 0x000c.
> knob --hnd 0x0b
[*] No key size available.
     - Did you already negotiate an encrypted connection?
     - Did you choose the correct connection handle?
[*] [Connection Create initiated]
[*] [Connect Complete: Handle=0xb  Address=05:ea:f0:0d:ca:fe  status=Success]
> knob --hnd 0x0b
[*] HCI_Read_Encryption_Key_Size result for handle 0xb: 1
>  
```

From the output you have pasted here with the key size 28, I would guess that this is the error state where you did not yet negotiate an encrypted session. I saw the same value in my Wireshark. You need perform a pairing and then also have an established, encrypted connection for the KNOB attack to become effective.

Hope that helps :)

phatut, 2019-09-29:
hi, 

I'm running with Nexus5. using the pre-compiled bluetooth.default.so... all is working fine, but the wireshark (upgraded to 3.0.5) isn't parsing the HCI packet type 0x7. Seeing exactly what Aleberello reported in his first comment on this thread. 

any ideas as to why this is not working? 

[LMP_monitoring_not_working.pcapng.gz](https://github.com/seemoo-lab/internalblue/files/3666421/LMP_monitoring_not_working.pcapng.gz)

jiska2342, 2019-09-29:
Hi @phatut,

your log works perfectly fine on my test system. You might still have the Wireshark 2.6 plugin. I updated it to Wireshark 3 (and tested with 3.0.1) a few days ago:

https://github.com/seemoo-lab/h4bcm_wireshark_dissector/issues/1

If you can see packets with H4 type `0x07` you're almost good, it is just a dissector issue :)

phatut, 2019-09-30:
hi @jiska2342 
I'm trying to build the dissector but am getting the following err after 'make' 

root@box:~/h4bcm_wireshark_dissector/build# make 
[ 20%] Generating plugin.c
Generated plugin.c for h4bcm_wireshark_dissector.
Scanning dependencies of target h4bcm
[ 40%] Building C object CMakeFiles/h4bcm.dir/plugin.c.o
**/root/h4bcm_wireshark_dissector/build/plugin.c:7:10: fatal error: config.h: No such file or directory
 #include "config.h"**
          ^~~~~~~~~~
compilation terminated.
CMakeFiles/h4bcm.dir/build.make:68: recipe for target 'CMakeFiles/h4bcm.dir/plugin.c.o' failed
make[2]: *** [CMakeFiles/h4bcm.dir/plugin.c.o] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/h4bcm.dir/all' failed
make[1]: *** [CMakeFiles/h4bcm.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

phatut, 2019-09-30:
resolved.. had to change the build/plugin.c: 

- #include "config.h" 
+ #include "wireshark/config.h"

jiska2342, 2019-09-30:
@phatut thank you for the comment, I changed the file in the automatic generator script. For whatever reason, both variants work on my system.

Aleberello, 2019-09-30:
Hi @jiska2342, 

Edit: maybe I solved it, look below

--------------------------------------------
I'm doing my best to make the same passages you gave me but it continue to give me the same outputs.
This is what happens:
1) I use Android to pair the two devices.
I check that they are actually connected via the adb command `adb shell dumpsys bluetooth_manager`, which outputs:
```
Bonded devices:
    {MAC addr} [BR/EDR] Galaxy A5 (2017)
```

2) Simultaneously internalblue gives me output:
```
[Connection Create initiated]
[Connect Complete: Handle=0xc   Address={MAC addr}    status=Success]
[Disconnect Complete: Handle=0xc]
```
But is it normal for him to disconnect immediately after making the pairing?
I tried also with a more ancient device (Xperia J) but it behaves the same way.

3) If i try to send the `sendhcicmd 0x1408 0x0c` just before it disconnect i get always this response:
```
Encryption Key Size: 28 -> [/2f 0c 00 00]
```

4) I also tried to send `sendhcicmd 0x1408 0x0c` after having forced the connection via InternalBlue (using `connect {MAC addr}`) both with and without having previously paired via android the devices and it outputs me: 
```
Encryption Key Size: 0 -> [/2f 0c 00 1c]
or
Encryption Key Size: 28 -> [/2f 0c 00 00]
```

--------------------------------------------
Anyway, at this exact moment I was able to carry out the check:

you told me that `sendhcicmd 0x1408 0x0c` only works if there is an active encrypted connection, otherwise the command does not give me the key length (since none is used).
So if I'm not wrong the `connect` command from internalblue doesn't establish an encrypted connection [point 4] and when I pair the two devices the connection decays immediately afterwards as it is not used for any profile (such as data transfer or listening to music - A2DP) [point 2].

Understood this I tried to transfer a file between the two devices, the result?
**On older device (Xperia J, Bluetooth 2.1)**
- Pre KNOB -> Key Size = 16
- After KNOB -> Key Size = 1 ("yay!")

**On "newer" device (Galaxy A5 before August security patch)**
- Pre KNOB -> Key Size = 16
- After KNOB ->  `unable to transfer file` so i think that the Galaxy A5 refuses the negotiation (abort connection). [strange, I thought they had solved the problem after the August patche]

What do you think about it? I'm right?

Last thing, could you briefly explain to me what memory addresses the `KNOB_PoC` script overwrites?
```
# modify function lm_SendLmpEncryptKeySizeReq
patch = asm("mov r2, #0x1", vma=0x5AED0)  # connection struct key entropy
internalblue.patchRom(0x5AED0, patch)

# modify global variable for own setting
internalblue.writeMem(0x203797, '\x01')  # global key entropy
```
I can't understand the difference between the first and second instructions.

Thank you very much, I know that for you they are trivial things but for me they are very interesting but very complicated to understand. :)

jiska2342, 2019-09-30:
I don't know how and if Android solved the KNOB issue. While iOS shows a warning, I don't know Android counteracts and how. It might be that it just resets the connection, but that would be a bit strange. Overall, Bluetooth on Android (and Bluetooth in general) is so unstable, that I would rather expect a reset to happen because it's Bluetooth.

To be honest, I never used Bluetooth for file transfers successfully. I know that is supposed to work, but as iOS user, the only thing that works for me is AirDrop or simply email ;) AirDrop performs perfectly well when sending 200MB, since it uses Bluetooth for discovery and then does a Wi-Fi based direct transfer.

What I do know is that it is absolutely normal to get a disconnect immediately after a connect. If there is no action that keeps the two devices connected, the connection will be dropped within seconds. You can prevent this i.e. by setting up Bluetooth tethering / Internet sharing. This works well between iOS and Android devices as well as Android and Android devices. Moreover, connections to headsets, keyboards, etc. are kept established.

If you want to check if your connection was dropped because of KNOB, you can get a copy of the `btsnoop_hci.log` or connect Wireshark to your Android phone under test. The only way for Android to detect KNOB is to issue the very same HCI command after the connection was established. However, it might be easier to check first if you can actually maintain a connection without KNOB at all.

In my patch, I think it would technically be sufficient to overwrite the global key size entry (the second patch). However, just to be sure nothing is overwritten during connection setup, I additionally patch the LMP handler that sends this key size request.

Aleberello, 2019-10-08:
Hi @jiska2342,

thank you very much for your patience and for all the time you gave me. I can close the issue with satisfaction.
I hope this thread can also help other people.
Thanks, good evening and wishes for your research!
Alessandro

closed 2019-10-08 by Aleberello