seemoo-lab/internalblue #9: Cannot send LMP packet

allenjlw, 2019-10-09:
Hi,

It's a very capable framework and it's much easier to deal with the firmware and low-level Bluetooth packets.

It works well unless when I was using the framework to send an LMP packet to another device. I encountered some errors and couldn't send the packet.

I was using the Nexus 5 phone (Android 6.0.1) with the library from this repo `bluetooth.default.so`. I can connect to the chip from my laptop and run some commands (like `hexdump`, `trace add`). When I connected to a device and send an LMP packet, the following error occurred. I got an error with error status 2.

[image]

I captured all the traffics via `monitor start` with Wireshark and it could be accessed in the link:
https://drive.google.com/open?id=1oUN3fm3ln33NPhmZ-2NeufH-3Adc6fAx

I've also tried with the evaluation board CYW920735 and Raspberry Pi 3 Model B. I got error status 0x12 from both of these two devices.

What might be the problem?

Thanks in advance.

jiska2342, 2019-10-09:
Hi @allenjlw,

you can only send LMP packets when you are connected to another device. Overall, you performed the correct steps to do so, but the issue here is that you got a `Connect Complete` event with the status `Page Timeout`. This means that a connection was not established. Instead, the status `Success` is required.

For a connection to succeed, the other device needs to be within wireless range and it also needs to be connectable. If that is the case depends on the configuration. Most devices are always connectable if you know their MAC address, but this is not necessarily the case. Instead of typing `connect de.ad:ca:fe:be:ef` or similar, you can also establish a connection using the GUI of your Android smartphone.

The next issue is that without additional intents to keep a connection open, it will be closed almost immediately. To keep a connection open, you can do tethering, continuously run `l2ping`or similar.

Moreover, not all implementations of `sendlmp` will determine the connection handle correctly. On the Nexus 5, this is done automatically by an additional script in the background if you do not provide it. Ideally, you also pass the connection handle (in your case `0x0c`, as indicated by the `Connect Complete` event) as parameter. This might be the reason why you got a different error status on the CYW20735 evaluation board and Raspberry Pi 3.

Note that *IntenalBlue* has an additional script for the Nexus 5, which you can activate by entering `fuzzlmp` once before running `sendlmp` later. This way, you can also send invalid LMP commands.

allenjlw, 2019-10-10:
Hi @jiska2342 ,

Thanks so much for the quick reply.
I removed the `bcm4335c0.hcd` file from Nexus 5 to have a simpler firmware. However, Nexus 5 Bluetooth cannot work well without that patch. That's the reason why the phone cannot connect to the devboard or Raspberry Pi.

It's working now. I can send the lmp packets.

Another minor issue is that the `h4bcm` plugin does not work for my Wireshark version which is 2.6. The error message is that `h4bcm.so does not have plugin_release symbol` when loading the plugin. I temporarily patched this by adding
```
WS_DLL_PUBLIC_DEF const gchar plugin_release[] = "2.6";
```
after line 133 in `tools/make-plugin-reg.py` file.

closed 2019-10-10 by allenjlw

jiska2342, 2019-10-10:
Thanks for closing the issue, @allenjlw. Two more minor additions to this topic:

`.hcd` files are not full of n-day security fixes. The sad truth is that they contain basic fixes for the chip to work at all that appear between chip production and release of a device with that chip. They are mainly to provide interopability between the Broadcom chip and other chips, as well as the chip and the operating system. The Nexus 5 is not the only device where Bluetooth does not work if you do not have a proper `.hcd` file installed, which is the reason why there are some repositories for these files on the Internet.

I need to try if your fix breaks the update for Wireshark 3 or not. The version that worked with Wireshark 2.6 is in the commit `f2a4e142ec586ae57bc7f517501e772ab14f6e4d`.

allenjlw, 2019-10-10:
Thanks for making the `.hcd` file more clear and the plugin that works with Wireshark 2.6.

Yeah, I realized that the `.hcd` file is not only for security patches but mainly for adding new functionalities. I don't have Wireshark 3 so I'm not sure if the patch works with Wireshark 3. It would be great if you can test that.

Thanks again.