seemoo-lab/internalblue #12: Uanble to connect to Evaluation Board CYW20735

wuruoyu, 2020-01-09:
Hi, 

I have an issue connecting to the Evaluation board CYW20735. The jumpers on the board are set as default and the demo application from WICED can be successfully downloaded to board. But it seems that Internalblue does not identify it as Cypress chipset. I print out the vendor ID, and it is 29 (0x1D). The screenshot is attached below:
[image]

Do I do something wrong or forget to set something? I appreciate your help in advance!

Respectfully,
Ruoyu

jiska2342, 2020-01-09:
Hi Ruoyu :)

29 is Qualcomm (see https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers/), so you're likely using your internal Bluetooth chip instead. The evaluation board's MAC address always starts with 20:73:... and so on, similar to its version.

The evaluation board itself will not work out of the box, you need to follow the steps listed [here](https://github.com/seemoo-lab/internalblue/tree/master/linux_bluez). Often this will only work after a few tries, but what always works for me is to use the `ClientControl` binary in WICED Studio and open and close the serial port, and then attach with `btattach`.

closed 2020-01-09 by jiska2342

jmr123456-beep, 2020-12-18:
Hi wuruoyu, would you teach me how to use CYW20735 with Internalblue? I am fresh man in this field. I don't understand if there is no ADB in CYW, is it really possible to communicate with Internalblue ? Thank you.