seemoo-lab/internalblue #14: Hexdump the 0x260000 section from Rasp Pi

wuruoyu, 2020-01-30:
Hi, 

The Bluetooth stack crashes when we do hexdump from Rasp Pi starting from 0x260000. The log:

```
[*] No adb devices found.
[*] HCI device: hci0  [B8:27:EB:0F:72:E3]  flags=13<UP RUNNING PSCAN>
[*] Connected to hci0
[*] Chip identifier: 0x2209 (001.002.009)
[*] Using fw_0x2209.py
[*] Loaded firmware information for BCM43430A1.
[*] Try to enable debugging on H4 (warning if not supported)...
[!] _sendThreadFunc: Sending to socket failed, reestablishing connection.
    With HCI sockets, some HCI commands require root!
> 
> hexdump -l 16 0x260000
[!] Received Evaluation Stack-Dump Event (contains 10 registers):
[!] pc: 0x0006b4e4   lr: 0x0006b575   sp: 0x0021f728   r0: 0x00260000   r1: 0x00000010
    r2: 0x0021f76e   r3: 0x80000000   r4: 0x00000000   r5: 0x002117b4   r6: 0x0001c025
[!] _sendThreadFunc: No response from the firmware.
[!] sendHciCommand: waiting for response timed out!
[!] readMem: No response to readRAM HCI command! (read_addr=260000, len=10)
[!] _sendThreadFunc: No response from the firmware.
[!] sendHciCommand: waiting for response timed out!
[!] readMem: No response to readRAM HCI command! (read_addr=260000, len=10)
[!] _sendThreadFunc: No response from the firmware.
[!] sendHciCommand: waiting for response timed out!
[!] readMem: No response to readRAM HCI command! (read_addr=260000, len=10)
[!] _sendThreadFunc: No response from the firmware.
[!] sendHciCommand: waiting for response timed out!
[!] readMem: No response to readRAM HCI command! (read_addr=260000, len=10)
[!] readMem: failed!
[!] Command failed: hexdump -l 16 0x260000
> exit
[*] Shutdown complete.
[*] Goodbye

```

Thank you!

jiska2342, 2020-01-30:
So, what exactly are you trying to do?

Not all memory is mapped. The `HCI_Read_RAM` command simply crashes on older devices, and abstracts a bit of this on newer devices. The stack dump that you get is exactly in that helper. So, if you compare the function from the LR at `0x0006b575`, you can spot obvious similarities with the `bt_boot_hci_Read_RAMHelper` of some symbolized firmware.

There is also some memory that is mapped but needs 4 byte aligned readout. Newer firmware versions do this, older ones just crash. For the *Nexus 5*, @demantz did implement a special hook that does 4 byte aligned copies. So if you are sure that `0x260000` is mapped, you might need to port this.

Anyway, this is nothing that *InternalBlue* aims to solve. *Frankenstein* will come with some ARM hooks written by @bolek42 that poke for a complete memory map. And even this hook might crash in a couple of super weird mapped hardware registers.

wuruoyu, 2020-02-01:
Hi, thank you for your quick response. 
In fw2209 which the Rasperry Pi 3 is using, there is memory secion:
`MemorySection(0x260000, 0x268000, True , False)`
So when we dump the fw from Rasperry Pi, this section will be dumped. 

But we find that when dumping this section, the BT chip will crash. And align flag does not work here. Is there something wrong with what we are doing? Or this section just can not be dumped ?

Thank you!

jiska2342, 2020-02-03:
All right, I removed that memory section. I also had another pending commit for the Raspberry Pi 3. I currently try to not to push too many things as we are migrating to Python 3.

Which sections can be dumped and wich cannot is a bit random. A few of them might also depend on initialization. Basically, without some special *Frankenstein* scripts and just the simple Broadcom `HCI_Read_RAM` handler, it is a bit random what you get.

closed 2020-02-03 by jiska2342