seemoo-lab/internalblue #17: Problems with blueZ on Raspberry pi 3b+

r00tb3, 2020-03-24:
Hi all, I've setup bluez with Raspbian OS on top of Raspberry Pi 3 b+ but when I run the internal blue script I get the following errors:

Using python2:

```
$ sudo python2 internalblue/cli.py
Traceback (most recent call last):
  File "internalblue/cli.py", line 36, in <module>
    import internalblue.utils.pwnlib_wrapper as pwnlib
ImportError: No module named internalblue.utils.pwnlib_wrapper
```

Using Python3

```
$ sudo python3 internalblue/cli.py
[!] Pwntools does not support 32-bit Python.  Use a 64-bit release.
Traceback (most recent call last):
  File "internalblue/cli.py", line 41, in <module>
    from .adbcore import ADBCore
SystemError: Parent module '' not loaded, cannot perform relative import
```

I've reinstalled pwntools by running:

```
internalblue# pip3 install .
Processing /x/y/z/internalblue
  Requirement already satisfied (use --upgrade to upgrade): internalblue==0.4 from file:///home/pi/Desktop/internalblue in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied: future in /usr/local/lib/python3.5/dist-packages (from internalblue==0.4)
Requirement already satisfied: pwntools>=4.0.1 in /usr/local/lib/python3.5/dist-packages (from internalblue==0.4)
Requirement already satisfied: pyelftools in /usr/local/lib/python3.5/dist-packages (from internalblue==0.4)
Requirement already satisfied: tox>=1.8.1 in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: pip>=6.0.8 in /usr/lib/python3/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: capstone>=3.0.5rc2 in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: python-dateutil in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: pyserial>=2.7 in /usr/lib/python3/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: psutil>=3.3.0 in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: intervaltree>=3.0 in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: requests>=2.0 in /usr/lib/python3/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: pygments>=2.0 in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: sortedcontainers in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: unicorn>=1.0.2rc1 in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: pysocks in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: mako>=1.0.0 in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: paramiko>=1.15.2 in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: ropgadget>=5.3 in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: packaging in /usr/local/lib/python3.5/dist-packages (from pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: virtualenv>=16.0.0 in /usr/local/lib/python3.5/dist-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: toml>=0.9.4 in /usr/local/lib/python3.5/dist-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
Collecting six<2,>=1.14.0 (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
  Downloading https://files.pythonhosted.org/packages/65/eb/1f97cb97bfc2390a276969c6fae16075da282f5058082d4cb10c6c5c1dba/six-1.14.0-py2.py3-none-any.whl
Requirement already satisfied: filelock<4,>=3.0.0 in /usr/local/lib/python3.5/dist-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: pluggy<1,>=0.12.0 in /usr/local/lib/python3.5/dist-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: py<2,>=1.4.17 in /usr/local/lib/python3.5/dist-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: importlib-metadata<2,>=0.12; python_version < "3.8" in /usr/local/lib/python3.5/dist-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: MarkupSafe>=0.9.2 in /usr/lib/python3/dist-packages (from mako>=1.0.0->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: bcrypt>=3.1.3 in /usr/local/lib/python3.5/dist-packages (from paramiko>=1.15.2->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: cryptography>=2.5 in /usr/local/lib/python3.5/dist-packages (from paramiko>=1.15.2->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: pynacl>=1.0.1 in /usr/local/lib/python3.5/dist-packages (from paramiko>=1.15.2->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: pyparsing>=2.0.2 in /usr/local/lib/python3.5/dist-packages (from packaging->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: importlib-resources<2,>=1.0; python_version < "3.7" in /usr/local/lib/python3.5/dist-packages (from virtualenv>=16.0.0->tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: distlib<1,>=0.3.0 in /usr/local/lib/python3.5/dist-packages (from virtualenv>=16.0.0->tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: appdirs<2,>=1.4.3 in /usr/local/lib/python3.5/dist-packages (from virtualenv>=16.0.0->tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: zipp>=0.5 in /usr/local/lib/python3.5/dist-packages (from importlib-metadata<2,>=0.12; python_version < "3.8"->tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: cffi>=1.1 in /usr/local/lib/python3.5/dist-packages (from bcrypt>=3.1.3->paramiko>=1.15.2->pwntools>=4.0.1->internalblue==0.4)
Requirement already satisfied: pycparser in /usr/local/lib/python3.5/dist-packages (from cffi>=1.1->bcrypt>=3.1.3->paramiko>=1.15.2->pwntools>=4.0.1->internalblue==0.4)
Installing collected packages: six
  Found existing installation: six 1.12.0
    Not uninstalling six at /usr/lib/python3/dist-packages, outside environment /usr
Successfully installed six-1.14.0
```

Also, I've installed and tested hcitool and bluetoothctl it worked perfectly... from [here](https://github.com/seemoo-lab/internalblue/tree/master/linux_bluez)

As far as enabling [diagnostics](https://github.com/seemoo-lab/internalblue/tree/master/linux_bluez#diagnostics) is concerned there's no `vendor_diag` in `hci0` on raspbian OS.

What am I doing wrong here? Can someone shed some light on this?

Thanks for reading.

fmagin, 2020-03-24:
Did you install internalblue via pip or are you just trying to run the script from the git checkout?

From a git checkout:
```bash
git clone https://github.com/seemoo-lab/internalblue
cd internalblue
pip install ./
python -m internalblue.cli # or just 'internalblue'
```

Simple install via pip (for now, until we will publish to PyPI at some point):
```bash
pip install https://github.com/seemoo-lab/internalblue/archive/master.zip
python -m internalblue.cli # or just 'internalblue'
```

jiska2342, 2020-03-24:
I just installed *InternalBlue* on a plain up-to-date Raspbian yesterday. Before running `pip3 install .`, I installed the following packages from the repositories: `apt-get install git python3-setuptools binutils-arm-none-eabi adb python3-pip python3-dev gcc libffi-dev`.

Note that your operating system should be up to date. Your log output above shows `/usr/local/lib/python3.5/`. *InternalBlue* does not work on Python 3.5. I had that issue yesterday, when I tried to install it on Ubuntu 16.04 LTS, because I wasn't the one who set up the Raspberry Pi and didn't know about the version running on it. Even if you resolve all pip dependencies, that won't help ;)

In case you cannot upgrade, check out the older Python 2 tag: https://github.com/seemoo-lab/internalblue/releases/tag/python2

jiska2342, 2020-03-24:
And another comment I forgot about :D

The `vendor_diag` feature in *BlueZ* is weird. They screwed up things in their drivers, meaning that it sometimes appears and sometimes not. All Broadcom chips support `vendor_diag`, but depending on if that chip is attached via the USB or via the UART sub-driver, the `vendor_diag` might be listed or not. @psy once looked into that, but the code is an undocumented mess, so we didn't dig deeper to solve that issue.

vaibhvbedi, 2020-03-31:
Can you please tell me which raspberry pi version are you using?? I tried with raspberry pi version 4 but not able to capture the pairing packets.

jiska2342, 2020-03-31:
@r00tb3 I'm closing this issue because @fmagin added a dependency that marks Python 3.5 as incompatible. The Python 3 version of *InternalBlue* added some type checks etc. that cannot be ported back to Python 3.5, so getting it to work on Python 3.5 is a won't fix. Please note that Python 3.5 was released in 2015, and Python 3.6 in 2016, so supporting Python 3.6 (which we do) is already a lot of backwards compatibility.

closed 2020-03-31 by jiska2342

r00tb3, 2020-04-05:
Hi @jiska2342, thanks for the comments I've upgraded to Python 3.6.8 but I'm struggling to make internalblue run over bluez.

I'm facing the following error:

```
internalblue $ sudo internalblue
[!] Pwntools does not support 32-bit Python.  Use a 64-bit release.
[*] Checking for new versions of pwntools
    To disable this functionality, set the contents of /root/.pwntools-cache-3.5/update to 'never'.
[*] You have the latest version of Pwntools (4.0.1)
Traceback (most recent call last):
  File "/usr/local/bin/internalblue", line 11, in <module>
    load_entry_point('internalblue==0.4', 'console_scripts', 'internalblue')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.5/dist-packages/internalblue/cli.py", line 41, in <module>
    from .adbcore import ADBCore
  File "/usr/local/lib/python3.5/dist-packages/internalblue/adbcore.py", line 44
    self.hciport: Optional[int] = None  # hciport is the port number of the forwarded HCI snoop port (8872). The inject port is at hciport+1
                ^
SyntaxError: invalid syntax
```

r00tb3, 2020-04-05:
Hi @jiska2342 , I've upgraded to Python 3.6 but I'm facing another issue now.

```
internalblue $ sudo pip3.6 install ./
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/Desktop/internalblue
Requirement already satisfied: pwntools>=4.0.1 in /usr/local/lib/python3.6/site-packages (from internalblue==0.4) (4.0.1)
Requirement already satisfied: pyelftools in /usr/local/lib/python3.6/site-packages (from internalblue==0.4) (0.26)
Requirement already satisfied: future in /usr/local/lib/python3.6/site-packages (from internalblue==0.4) (0.18.2)
Requirement already satisfied: pygments>=2.0 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (2.6.1)
Requirement already satisfied: psutil>=3.3.0 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (5.7.0)
Requirement already satisfied: pyserial>=2.7 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (3.4)
Requirement already satisfied: mako>=1.0.0 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (1.1.2)
Requirement already satisfied: requests>=2.0 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (2.23.0)
Requirement already satisfied: paramiko>=1.15.2 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (2.7.1)
Requirement already satisfied: capstone>=3.0.5rc2 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (4.0.1)
Requirement already satisfied: intervaltree>=3.0 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (3.0.2)
Requirement already satisfied: unicorn>=1.0.2rc1 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (1.0.2rc2)
Requirement already satisfied: tox>=1.8.1 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (3.14.6)
Requirement already satisfied: pysocks in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (1.7.1)
Requirement already satisfied: python-dateutil in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (2.8.1)
Requirement already satisfied: pip>=6.0.8 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (20.0.2)
Requirement already satisfied: packaging in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (20.3)
Requirement already satisfied: sortedcontainers in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (2.1.0)
Requirement already satisfied: ropgadget>=5.3 in /usr/local/lib/python3.6/site-packages (from pwntools>=4.0.1->internalblue==0.4) (6.2)
Requirement already satisfied: MarkupSafe>=0.9.2 in /usr/local/lib/python3.6/site-packages (from mako>=1.0.0->pwntools>=4.0.1->internalblue==0.4) (1.1.1)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.6/site-packages (from requests>=2.0->pwntools>=4.0.1->internalblue==0.4) (2019.11.28)
Requirement already satisfied: idna<3,>=2.5 in /usr/local/lib/python3.6/site-packages (from requests>=2.0->pwntools>=4.0.1->internalblue==0.4) (2.9)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /usr/local/lib/python3.6/site-packages (from requests>=2.0->pwntools>=4.0.1->internalblue==0.4) (1.25.8)
Requirement already satisfied: chardet<4,>=3.0.2 in /usr/local/lib/python3.6/site-packages (from requests>=2.0->pwntools>=4.0.1->internalblue==0.4) (3.0.4)
Requirement already satisfied: bcrypt>=3.1.3 in /usr/local/lib/python3.6/site-packages (from paramiko>=1.15.2->pwntools>=4.0.1->internalblue==0.4) (3.1.7)
Requirement already satisfied: pynacl>=1.0.1 in /usr/local/lib/python3.6/site-packages (from paramiko>=1.15.2->pwntools>=4.0.1->internalblue==0.4) (1.3.0)
Requirement already satisfied: cryptography>=2.5 in /usr/local/lib/python3.6/site-packages (from paramiko>=1.15.2->pwntools>=4.0.1->internalblue==0.4) (2.9)
Requirement already satisfied: py<2,>=1.4.17 in /usr/local/lib/python3.6/site-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4) (1.8.1)
Requirement already satisfied: filelock<4,>=3.0.0 in /usr/local/lib/python3.6/site-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4) (3.0.12)
Requirement already satisfied: toml>=0.9.4 in /usr/local/lib/python3.6/site-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4) (0.10.0)
Requirement already satisfied: importlib-metadata<2,>=0.12; python_version < "3.8" in /usr/local/lib/python3.6/site-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4) (1.6.0)
Requirement already satisfied: six<2,>=1.14.0 in /usr/local/lib/python3.6/site-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4) (1.14.0)
Requirement already satisfied: pluggy<1,>=0.12.0 in /usr/local/lib/python3.6/site-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4) (0.13.1)
Requirement already satisfied: virtualenv!=20.0.0,!=20.0.1,!=20.0.2,!=20.0.3,!=20.0.4,!=20.0.5,!=20.0.6,!=20.0.7,>=16.0.0 in /usr/local/lib/python3.6/site-packages (from tox>=1.8.1->pwntools>=4.0.1->internalblue==0.4) (20.0.16)
Requirement already satisfied: pyparsing>=2.0.2 in /usr/local/lib/python3.6/site-packages (from packaging->pwntools>=4.0.1->internalblue==0.4) (2.4.6)
[... cut ...]
  File "/home/pi/Desktop/internalblue/internalblue/cli.py", line 273, in internalblue_cli
    devices.extend(connection_method.device_list())
  File "/home/pi/Desktop/internalblue/internalblue/hcicore.py", line 178, in device_list
    for dev in self.getHciDeviceList():
  File "/home/pi/Desktop/internalblue/internalblue/hcicore.py", line 83, in getHciDeviceList
    s = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_RAW, socket.BTPROTO_HCI)
AttributeError: module 'socket' has no attribute 'AF_BLUETOOTH'
```
What am I missing here?

Regards.

jiska2342, 2020-04-05:
If you don't have the socket type `AF_BLUETOOTH` something else on your system is fundamentally wrong. Are you sure you are on a Raspberry Pi 3B+? What is the output of `hciconfig` on your device?

r00tb3, 2020-04-05:
> If you don't have the socket type `AF_BLUETOOTH` something else on your system is fundamentally wrong. Are you sure you are on a Raspberry Pi 3B+? What is the output of `hciconfig` on your device?

```
$ sudo hciconfig 
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:AA:E9:9D  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:813 acl:0 sco:0 events:53 errors:0
	TX bytes:2524 acl:0 sco:0 commands:53 errors:0

```