seemoo-lab/internalblue #18: core.py - Python3 problem

bobwilmes, 2020-03-25:
I am getting a python error  (TypeError: can't concat str to bytes)
at line 1196 in core..py. Can you please help me fix ?
I am running a Nexus5 (Android 6.0.1) pairded to an iPhone X
[image]

bobwilmes@bobwilmes:~$ adb devices
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached
0759f01913d7c4c2	device

bobwilmes@bobwilmes:~$ cd internalblue
bobwilmes@bobwilmes:~/internalblue$ ls
android_bluetooth_stack  dummymemdump.bin  examples  internalblue  internalblue_thesis_dennis_mantz.pdf  ios-internalblued  linux_bluez  macos-framework  mypy.ini  README.md  setup.py  tests
bobwilmes@bobwilmes:~/internalblue$ ls examples/nexus5
BLE_Reception_PoC.py  CVE_2018_5383_Invalid_Curve_Attack_PoC.py  KNOB_PoC.py  LMP_MAC_Address_Filter.py  NiNo_PoC.py
bobwilmes@bobwilmes:~/internalblue$ sudo python3 examples/nexus5
[sudo] password for bobwilmes: 
/usr/bin/python3: can't find '__main__' module in 'examples/nexus5'
bobwilmes@bobwilmes:~/internalblue$ sudo python3 examples/nexus5/KNOB_PoC.py
[*] Found multiple adb devices
[*] Connected to 0759f01913d7c4c2
[*] Chip identifier: 0x6109 (003.001.009)
[*] Using fw_0x6109.py
[*] Loaded firmware information for BCM4335C0.
[*] Try to enable debugging on H4 (warning if not supported)...
[*] Installing patch which ensures that send_LMP_encryptoin_key_size_req is always len=1!
[*] patchRom: Choosing next free slot: 113
Traceback (most recent call last):
  File "examples/nexus5/KNOB_PoC.py", line 42, in <module>
    internalblue.writeMem(0x203797, '\x01')  # global key entropy
  File "/usr/local/lib/python3.6/dist-packages/internalblue/core.py", line 1196, in writeMem
    p32(write_addr) + data[byte_counter : byte_counter + blocksize],
TypeError: can't concat str to bytes
bobwilmes@bobwilmes:~/internalblue$

fmagin, 2020-03-25:
That script was simply outdated and not properly ported to Python 3 yet. Could you please test if https://github.com/seemoo-lab/internalblue/tree/fix_examples works for you?

bobwilmes, 2020-03-25:
Hi Florian,
I cloned the repository and re-ran the test.
The problem is line 1196 in the core.py module in the
internalblue/internalblue/core.py

Python reports:
[*] patchRom: Reusing slot for address 0x5aed0: 113
Traceback (most recent call last):
  File "examples/nexus5/KNOB_PoC.py", line 42, in <module>
    internalblue.writeMem(0x203797, '\x01')  # global key entropy
  File "/usr/local/lib/python3.6/dist-packages/internalblue/core.py", line
1196, in writeMem
    p32(write_addr) + data[byte_counter : byte_counter + blocksize],
TypeError: can't concat str to bytes

Can you fix this in core.py ? I don't know enough Python but it appears to
be the string concat doesn't work under Python3

Regards,
Bob Wilmes
Omaha, Nebraska, USA

bobwilmes, 2020-03-25:
Hi Florian,
i did some research and there appear to be
three lines in core.py that have the same
string concatenation problem.

Line 318, 867 and 1169

Regards , Bob Wilmes

fmagin, 2020-03-26:
You aren't running the updated version of the script. The problem is the line

```python
internalblue.writeMem(0x203797, '\x01')
```
Which passes a 'str' to `writeMem`, but in Python 3 the argument to `writeMem` must be `bytes`.
The fix is simply to change it to
```python
internalblue.writeMem(0x203797, b'\x01')
```

That is changed in the branch https://github.com/seemoo-lab/internalblue/blob/fix_examples/examples/nexus5/KNOB_PoC.py#L40 already but your stack trace shows that you aren't running the updated version.

bobwilmes, 2020-03-26:
Thank you Florian - this fixed it!

Regards,
Bob WIlmes

fmagin, 2020-03-28:
Thanks for the report!

closed 2020-03-28 by fmagin