seemoo-lab/internalblue #19: Raspberry pi Version

vaibhvbedi, 2020-03-31:
Hi,
I would like to know which raspberry pi version is comfortable for running the internal Blue. Because I tried with the raspberry pi version 4 and the latest raspbian image, but I am not able to capture the pairing packets.

Thanks in advance.

jiska2342, 2020-03-31:
The Raspberry Pi 3 and Zero W have the *BCM43430A1* chip, while the Raspberry Pi 3+ and 4 have the *BCM4345C0* chip. Firmwares are documented here: https://github.com/seemoo-lab/internalblue/blob/master/internalblue/fw/README.md

*BlueZ* most likely gets the `vendor_diag` option wrong if you don't see LMP. In case that LMP is what you mean with pairing messages? If `/sys/kernel/debug/bluetooth/hci0/vendor_diag` is missing, this is a *BlueZ* issue and does not have to do anything with *InternalBlue*.

vaibhvbedi, 2020-03-31:
Hi, I am using the raspbian OS version of February 2020. But this file is missing /sys/kernel/debug/bluetooth/hci0/vendor_diag is missing.
Can you please tell me which raspbian OS version is compatible with the internal blue.

closed 2020-03-31 by vaibhvbedi