seemoo-lab/internalblue #20: sendlmp: error: argument opcode: invalid auto_int value: '01'

SuperSadGuy23, 2020-04-04:
I have installed internalblue on Nexus5 and connected it to a device. But when I am giving 
sendlmp 01 -d 02
I am getting this weird error:
sendlmp: error: argument opcode: invalid auto_int value: '01'
Any helps please?

fmagin, 2020-04-04:
Ok, this is a really weird behavior of the ArgumentParser.
`parser.add_argument("opcode", type=auto_int, help="Number of the LMP opcode")` handles the argument, `01` in your case, where auto_int is just `int(x, 0)`. And for some reason `int('01', 0)` isn't valid:
```
In [2]: int('01', 0)                                                                                                                                           
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-2-92c2da78664f> in <module>
----> 1 int('01', 0)

ValueError: invalid literal for int() with base 0: '01'

```

The solution is simply to call it like `sendlmp 1 -d 02`, `02` is being parsed by `unhexlify` so this is no problem.

@jiska2342 Does this warrant a fix? Might be something that should be taken into account when the Command Line is refactored.

jiska2342, 2020-04-04:
Hi there,

I'm currently fixing this. The `sendlmp` command works best if also provided with a connection handle. If you do not enable tethering with the target device or audio streaming etc., you need to be fast, otherwise, the connection will be closed quite fast. This is why the Nexus 5 PoC for CVE-2018-19860 patches the connect routine. If everything works out, you can now use it as follows with my most recent git commit:

```
# No connection established: it fails
> sendlmp 35 -d 01020304050607 --conn_handle 12
[*] Sending op=35 data=\x03\x05\x07to connection handle=0x000c
[!] sendLmpPacket: got error status 0x02
[!] Command failed: sendlmp 35 -d 01020304050607 --conn_handle 12

# Connection established
[*] [Connection Create initiated]
[*] [Connect Complete: Handle=0xc  Address=cafebabe1337  status=Success]
> sendlmp 35 -d 01020304050607 --conn_handle 12
[*] Sending op=35 data=\x03\x05\x07to connection handle=0x000c
[*] [Disconnect Complete: Handle=0xc]
```

Note that the connection handle is still decimal and not hexadecimal. Also the data is sent correctly as you can observe in Wireshark with `btlmp || bthci_evt.opcode == 0xfc58 || bthci_cmd.opcode == 0xfc58`. You need the H4 BCM Wireshark plugin, to see it, though. So, a few more type issues to fix for me, sorry %)

When you're on a Nexus 5, I can also recommend to issue the command `fuzzlmp` once before using `sendlmp`, as it overwrites a length check in the original routine and lets you send invalid length LMP packets.

closed 2020-04-12 by jiska2342