seemoo-lab/internalblue #21: Connect command doesn't work (possible Python error)

KamicDemon, 2020-04-06:
Hello,
First I want to express my thanks to everyone that has helped support this awesome project. I'm so excited to try this, but I'm not very experienced so it's possible I'm making a lot of rookie mistakes.

I'm using a rooted Nexus 5 running Android 6.0.1. Build number is M4B30Z

I followed the instructions detailed in the android_bluetooth_stack README, using the pre-compiled android6_0_1. I did no building of my own. I ran the "git apply" command before pushing the .so file and running the other adb commands.

I installed the h4bcm_wireshark_dissector following the instructions on that git page.

I didn't encounter any errors while following those steps so I assume I did them right.

USB debugging and HCI snoop logs are both enabled.

When I run internalblue, I can open wireshark through "monitor start" after selecting my adb interface. I don't see any LMP traffic when initiating a pair request or scan from the Nexus5.

When I try to connect using the internalblue "connect" command I get errors and the program exits. Screenshots provided below.

Once again, I'm sorry if this is a total newbie mistake, I'm just not sure what's going on.

[image]
[image]

fmagin, 2020-04-06:
That is definitely some type error on our side, looking into it.

fmagin, 2020-04-06:
At least not an issue in current `master`.
On current `master` the address is converted with `addr = bytearray.fromhex(addr)` in `parse_bt_addr` at line 122.

That code hasn't changed since January 2019 either.
How did you install internalblue?

KamicDemon, 2020-04-06:
I cloned the repo, changed into the directory, and did
"sudo python3 setup.py install"
I think it was an issue with the machine I was using earlier... I made a fresh linux VM on another machine and gave it a try and it seems to be working a bit better. I can use the "connect" command without any issues, but I'm still not seeing LMP traffic.

I see lots of "Rcvd Unknown HCI Packet Type 0x07" messages, as you can see in the screenshot. I'm reading other issues to try and find a fix.

[image]

jiska2342, 2020-04-07:
Hi,

`Rcvd Unknown HCI Packet Type 0x07` is when your Wireshark dissector plugin doesn't work, that's the LMP packets. The dissector works in all versions of Wireshark 3.* I tried so far, but I encountered that the home directory changed in the most recent version of my machine and I needed to copy it to a different plugin subfolder to get it working.

I don't have any issues with the `connect` command, though, and it also works well in your screenshot.

fmagin, 2020-04-07:
My guess would be that `python3 setup.py install` doesn't upgrade packages, at least `pip install ./` doesn't unless you add `--upgrade`

KamicDemon, 2020-04-08:
I figured out the problem!
I double checked a few things in wireshark:
Help -> About Wireshark: Told me the exact version I was using
In that same "About Wireshark" Page there's a tab named "Folders" that lists both the global and personal plugin files Wireshark is sourcing its plugins from. I noticed that they were different than the one the h4bcm.so file was installed to.

Apparently, when running make install, a "plugins/3.0" folder was created to host the h4bcm.so file, but my Wireshark (Version 3.2.2) was sourcing from a "plugins/3.2" folder. The personal version of this directory didn't even exist, as i show in the attached screenshot.

Clicking on the link in the "About Wireshark" window prompted me to make the folder, which I did, and then copied and pasted the h4bcm.so file into **both** the global and personal plugin folders just to be safe - and it all works! Screenshots attached.

I hope this can help someone else if they encounter the same problem (although we've sort of drifted from the topic of in the issue title)

Closing the issue, thanks again for all the help :)
[image]
[image]
[image]
[image]

closed 2020-04-08 by KamicDemon