seemoo-lab/internalblue #23: Nexus 5: unable to locate symbol __android_log_error_write

mikeryan, 2020-04-12:
I am using a rooted Nexus 5 running an otherwise unmodified factory Android 6.0.1. I am testing with master (1abc8c7 as of this writing).

I followed the instructions to install the patched `bluetooth.default.so` from the `android6_0_1` subdirectory, but I am unable to enable Bluetooth after doing so. The following error appears in logcat output:

`dlopen failed: cannot locate symbol "__android_log_error_write" referenced by "/system/lib/hw/bluetooth.default.so"...`

The above error goes away if I replace `bluetooth.default.so` with the factory version, but obviously I can't use internalblue with the factory version.

jiska2342, 2020-04-12:
Hi @mikeryan,

just copying the `bluetooth.default.so` might not always work. Worst case, you need to recompile the Bluetooth stack as described in the **Build (AOSP)** section.

If you do not care about performance, you can still try to call *InternalBlue* with the `serial` option: `python3 -m internalbue.cli --serial`.

The `serial` option just netcats the `btsnoop_hci.log` as output and takes inputs that it directly writes to the Bluetooth serial device in `/dev/tty*`. This requires you to install the *BusyBox* APK, which ships with `netcat`. I just pushed one minor fix to `adbcore.py` as the `lsof` output looked a bit different on my *Nexus 5*. However, I do no longer have the original `bluetooth.default.so`, so I cannot try it myself. You can also make adaptions to this shellscripting in the function `_setupSerialSu`.

I'm using this hack for *Android 8, 9* and *10* ;)

mikeryan, 2020-04-12:
Thanks for the tip, I saw serial mode and wasn't sure if it'd work on the older OS. I'm still having some trouble getting that to work, but I can debug it on my own at this point. Meanwhile I'll kick off an AOSP build so I can just do it The Fast Way™. Will submit a PR with the `.so` if I can get that working.

P.S., the initial CCC talk on this material was great, thanks for that! Really looking forward to playing with this.

closed 2020-04-12 by mikeryan