seemoo-lab/internalblue #25: Disconnecting from the device

vaibhvbedi, 2020-04-15:
Hi,

I want to ask why I am getting disconnection all the time. It appears the internal blue is conflicting for some reason. and I removed the bcm4335c0.hcd file from Nexus 5 the as well as. while sending the Lmp packets it always shows command failed. Please look into this issue.

Thank you
[image]
[image]

[image]

jiska2342, 2020-04-15:
Removing the `.hcd` file will likely make Bluetooth not work at all on a *Nexus 5*. It contains necessary patches that are just there to make Bluetooth functional, `.hcd` files are most of the time not about security fixes. You need to restore it, otherwise you will experience a lot of weird behavior.

Moreover, a *Nexus 5* on *Andoird 6* will never keep a connection open for more than a few seconds if it does not have any purpose. You need to pair with a device that does music streaming or tethering, otherwise, the connection will be closed almost immediately. This is why the PoC for CVE-2018-19860 patches the connect command within the firmware.

If you experience problems with creating connections nonetheless, you might also have a `bluetooth.default.so` that is incompatible with your *Android* version, see [issue 23](https://github.com/seemoo-lab/internalblue/issues/23).

vaibhvbedi, 2020-04-16:
Hi Jiska,
Thanks for your prompt response.
I have again restored the .hcd file inside the firmware folder. and Nexus 5 Bluetooth is working fine.
I am using the Nexus 5 android version 7.1.2  and the Lineage OS version. and I am able to capture the pairing request on the Wireshark. but after the pairing, my device is suddenly getting disconnect. 
Look forward to hearing from you.

Thanks

[image]

jiska2342, 2020-04-16:
Pairing is completely independent. If there is no music or tethering etc., you will be disconnected regardless of pairing.

Tethering means that the other device has in fact some LTE connection or similar to share and is also authorized to share it. It doesn't always work. I got it working between an *iPhone SE* and a *Samsung Galaxy S8*, for example, with both of them being disconnected from Wi-Fi.

In your trace, you at least managed to establish some L2CAP/SDP connection to a remote Motorola. But already in packet 284, the remote Motorola (not your Nexus 5) is sending a request to disconnect. This indicates something is misconfigured on your Motorola. The error happens way before the packet that your marked in the trace.

All of this is totally normal Bluetooth behavior and does not have to do anything with *InternalBlue*.

vaibhvbedi, 2020-04-16:
Hi Jiska,
Really Thanks a lot for your help. Finally, internal blue is working. The only concern I have here, Is the target connection must be done from the initiating device to the target device which has tethering, not the other way around. means that we cannot start the connection from internal blue, but rather from the target.

Look forward to hearing from you.

Thanks

jiska2342, 2020-04-17:
Hi,

*InternalBlue* does not change anything in that process. In fact, it should be completely passive during all actions unless you do some fancy Assembly scripting or run some special commands. Even the LMP logging is some *Broadcom*-internal feature that is no weird hack at all.

I didn't play around with tethering a lot, but from a security perspective, it definitely makes sense to restrict Internet sharing to be only enabled when the device that provides the Internet connection initiates that process.

vaibhvbedi, 2020-04-18:
Hi Jiska,

Thanks for your reply. But still, I am not getting why we need to initiate the connection from the target device?
Like on the internal blue when I am sending the command connect **device Address** that time it's showing the disconnection. But if I am sending the request from my Motorola mobile to the nexus 5, I am able to connect with the device and send the lmp packets.

Please clarify my doubt about this.

jiska2342, 2020-04-23:
As said, you might need to initiate it from the Motorola due to security reasons.

Also, it is not clear to me what you mean by initiating a connection from *InternalBlue*. So, if you mean the *InternalBlue* command line `connect` command, that one just issues an HCI connection request. This might not be sufficient to initiate tethering, as this requires additional actions. There is no reason to issue anything else than a raw connect, *InternalBlue* does not do any additional magic in the background. It tries to interfere with the host stack as little as possible. If you use the Andoird GUI to initiate a connection it might work, but also no guarantee due to the same security aspect.

closed 2020-05-04 by jiska2342