seemoo-lab/internalblue #28: CLI interface bug

krviolent, 2020-05-28:
I've got this error unexpectedly, when I tapped "up arrow" key in cli while running InternalBlue. So this case should be properly handled...

Traceback:
[CRITICAL] Uncaught exception (can't concat str to bytes). Abort.
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/readline.py", line 398, in readline
    keymap.handle_input()
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/keymap.py", line 24, in handle_input
    self.send(key.get())
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/keymap.py", line 52, in send
    cb(self.trace)
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/readline.py", line 205, in history_prev
    set_buffer(history[history_idx], '')
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/readline.py", line 148, in set_buffer
    redisplay()
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/readline.py", line 120, in redisplay
    ret = complete_hook(buffer_left, buffer_right)
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/completer.py", line 45, in complete
    w = self._get_word(buffer_left)
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/completer.py", line 32, in _get_word
    if left[i] in self.delims:
TypeError: 'in <string>' requires string as left operand, not int

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/internalblue-0.4-py3.6.egg/internalblue/cli.py", line 81, in commandLoop
    pwnlib.term.readline.readline(prompt="> ").strip().decode("utf-8")
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/readline.py", line 412, in readline
    line = buffer_left + buffer_right + '\n'
TypeError: can't concat str to bytes

>  Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/readline.py", line 398, in readline
    keymap.handle_input()
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/keymap.py", line 24, in handle_input
    self.send(key.get())
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/keymap.py", line 52, in send
    cb(self.trace)
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/readline.py", line 205, in history_prev
    set_buffer(history[history_idx], '')
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/readline.py", line 148, in set_buffer
    redisplay()
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/readline.py", line 120, in redisplay
    ret = complete_hook(buffer_left, buffer_right)
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/completer.py", line 45, in complete
    w = self._get_word(buffer_left)
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/completer.py", line 32, in _get_word
    if left[i] in self.delims:
TypeError: 'in <string>' requires string as left operand, not int

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/internalblue", line 11, in <module>
    load_entry_point('internalblue==0.4', 'console_scripts', 'internalblue')()
  File "/usr/local/lib/python3.6/dist-packages/internalblue-0.4-py3.6.egg/internalblue/cli.py", line 163, in internalblue_entry_point
    return internalblue_cli(sys.argv[1:])
  File "/usr/local/lib/python3.6/dist-packages/internalblue-0.4-py3.6.egg/internalblue/cli.py", line 308, in internalblue_cli
    commandLoop(reference, init_commands=args.commands)
  File "/usr/local/lib/python3.6/dist-packages/internalblue-0.4-py3.6.egg/internalblue/cli.py", line 81, in commandLoop
    pwnlib.term.readline.readline(prompt="> ").strip().decode("utf-8")
  File "/usr/local/lib/python3.6/dist-packages/pwntools-4.2.0b0-py3.6.egg/pwnlib/term/readline.py", line 412, in readline
    line = buffer_left + buffer_right + '\n'
TypeError: can't concat str to bytes

fmagin, 2020-05-28:
Definitely a Python2 vs Python3 issue, though I am fairly sure that at some point during testing I used the up arrow key and didn't enocunter this. Might be an issue combined with pwntools. Will try to reproduce this on my machine.

fmagin, 2020-05-28:
Can reproduce this on my machine with Python 3.8 and with both pwntools `4.1.0b0` and `4.2.0b0`. It only happens though if you try going back into your history further than the first command of the current session I think. pwntools also breaks my PyCharm integration because it doesn't recognize it as a terminal anymore and than crashes because an argument that is never supposed to be used is `None`. So I guess the proper solution to this finally rip out pwntools for the CLI part and replace it  with https://cmd2.readthedocs.io/en/latest/. Which makes this an issue for @unixb0y .

jiska2342, 2020-05-28:
Yes, I had the same issue a while ago, fixed a few lines in `pwntools` in one place and then just got more type errors in other places within `pwntools` and decided it's too much effort ;)

unixb0y, 2020-10-02:
Hi everyone, do you still have this issue with pwntools 4.2.1? I remember this issue but I don't have it anymore, fresh installation of `internalblue` using the instructions in this repo for macOS. But usage of the latest pwntools should be platform independent due to our install script.

```
[...]
install_requires=["pwntools>=4.0.1", "pyelftools", "future"],
[...]
```

unixb0y, 2020-10-23:
Pwnlibs is not used anymore for the CLI so this issue is now definitely fixed.

closed 2020-10-23 by unixb0y