seemoo-lab/internalblue #35: Can controller to send arbitrary messages to the host？

starsdestinations, 2020-11-02:
I want the controller to send arbitrary messages to the host（controller is bluetooth‘s hardware，host is my computer），
such as sending L2cap connection messages before the HCI_ACL connection is established. Can internalblue achieve this function?

jiska2342, 2020-11-02:
Hi @starsdestinations,

*InternalBlue* does not have this function included by default, since it requires firmware patching. The patches would be different per device.

However, it is quite simple to add custom HCI/ACL commands with *InternalBlue* once you located the according handler within the firmware of the specific device you own. As of now, this requires writing custom patches in ARM assembler. You can find examples for adding custom HCI commands in various firmware versions if you take a look into the random number generator patches. Since L2CAP is ACL and not HCI, you might need to use a different function, though. Jan did some ACL patching for *Frankenstein* PoCs, which also led to the discovery of CVE-2020-0022, so it's definitely possible :)
Best,
Jiska

TornadoZzz, 2020-11-03:
Hi~ @jiska2342 
Oh, this issue is interesting. Has this function done if the device that I used is Nexus 5? Or can you point out the shortcut or the specific part in Nexus 5 according to your research about Nexus 5?
Sincerely

jiska2342, 2020-11-03:
Hi @TornadoZzz,

as of now, I would recommend using one of the *Cypress* development kits (*CYW20819* ideally, but *CYW20719* and *CYW20735* should also work well). The `patch.elf` files in *WICED Studio 6.2/6.4* leak function names, which makes patching much easier. As of now, you need to write patches in assembler, though. We started integrating things into *Nexmon*, and there is an older `bluetooth-wip` branch, but there are some things that changed in newer firmware versions within the patching mechanism that still need to be fixed and so on... *Frankenstein* is also capable to create patches specific for the *Cypress* development boards in C, though :)

... we're planning to release some tooling for this soon (read soon as in a few months), which includes function names, arguments, and a lot of data types, and all the tooling to write patches in C.

Best,
Jiska

starsdestinations, 2020-11-05:
Hi @jiska2342 ,
Thanks for your guidance, I will try it.

jiska2342, 2020-11-05:
Hi @starsdestinations,

just one more question if you want to answer it ;) The only application that comes to my mind when talking about sending custom L2CAP data from the controller to the host would be fuzzing. However, instead of modifying the firmware, you can also modify the UART/USB driver or the Bluetooth daemon on the host, which is much easier than this.

Sending L2CAP messages originating from the firmware itself is something you might need when you're writing an exploit that escalates into the host. Yet, to explore L2CAP escalations into the host with fuzzing or similar techniques, you don't need to modify the firmware.

Best,
Jiska

closed 2021-03-04 by jiska2342