seemoo-lab/internalblue #40: TypeError: a bytes-like object is required, not 'int'

jurek4321, 2021-01-15:
**Describe the bug**

**Hardware and operating system**

**To Reproduce**

**Logs or screenshots**

**Additional context**

jurek4321, 2021-01-16:
I found that the problem is with internalblue/utils/packing.py so i updated u8 function to this:
def u8(num, endian: str = ''):
    print(num)
    if endian.lower() == 'big':
        return struct.unpack('>B', num)[0]
    elif endian.lower() == 'little':
        return struct.unpack('<B', num)[0]
    if (type(num) == int):
        num = bytes([num])
    return struct.unpack('B', num)[0]
    
    but i often get errors like this on android logcat:
    01-16 05:46:00.951 15136 15169 E bt_hci  : command_timed_out hci layer timeout waiting for response to a command. opcode: 0x41d
01-16 05:46:00.952 15136 15169 E %s restarting the bluetooth process.: command_timed_out
01-16 05:46:00.952 15136 15169 I %s      : ssr_cleanup
01-16 05:46:00.953 15136 15169 E bt_hci  : hci_cmd_timeout: SOC Status is reset
01-16 05:46:00.953 15136 15169 E bt_hci  :  

and after this bluetooth is restarting.

jiska2342, 2021-01-16:
Hi @jurek4321,

definitely looks as if something goes wrong there. To debug if it is the `bluetooth.default.so` I can recommend just using plain Wireshark. Usually, if Android/Lineage loads the module correctly, you should see Android debug interfaces in Wireshark even if not calling it via InternalBlue, and you can select the BT Snoop Log. Looks like this on my Nexus 5:
[image]
You might need to turn off/on Bluetooth once after a reboot before it works.

Once you connect with InternalBlue, you should see a version command (because that's the first thing InternalBlue does to determine the correct firmware file), and you should also see the contents of the packet when or before it breaks.

@unixb0y did some changes to the `adb` part lately but AFAIK he tested everything on the Nexus 5.

If you see stuff in Wireshark but not in InternalBlue, I can do some debugging these days and check if the new `adb` implementation works on both macOS and Linux hosts.

Best,
Jiska

jiska2342, 2021-01-16:
Ah, and maybe one thing I should add:

If InternalBlue gets the correct firmware version answer, it will send a command starting with `0x7`, which enables diagnostics, instead of `0x1`, which are the usual HCI commands. Sending `0x7` might still succeed, but if the `bluetooth.default.so` driver is patched incorrectly, it might later on fail to parse the diagnostic information and just crash somewhen during parsing packets. That might be the error, since you're actually getting a crash within the Bluetooth daemon. But it might also be InternalBlue sending something completely unexpected.

jurek4321, 2021-01-16:
Thanks for replay, how i can get correct patch to build bluetooth stack for z3c?

jurek4321, 2021-01-17:
I tried to build it for z3c but when i run "breakfast z3c" i get following error:
repo sync has finished successfully.
Looking for dependencies in device/sony/common
Dependencies file not found, bailing out.
Looking for dependencies in device/qcom/common
Dependencies file not found, bailing out.
Looking for dependencies in hardware/sony/thermanager
Dependencies file not found, bailing out.
Looking for dependencies in hardware/sony/macaddrsetup
Dependencies file not found, bailing out.
Looking for dependencies in external/stlport
Dependencies file not found, bailing out.
Looking for dependencies in hardware/broadcom/fm
Dependencies file not found, bailing out.
Done
build/core/product_config.mk:254: *** _nic.PRODUCTS.[[device/sony/z3c/lineage.mk]]: "vendor/sony/msm8974-common/msm8974-common-vendor.mk" does not exist.  Stop.
build/core/product_config.mk:254: *** _nic.PRODUCTS.[[device/sony/z3c/lineage.mk]]: "vendor/sony/msm8974-common/msm8974-common-vendor.mk" does not exist.  Stop.

** Don't have a product spec for: 'lineage_z3c'
** Do you have the right repo manifest?
How to fix it?

jiska2342, 2021-01-17:
Hi @jurek4321,

building the module properly is somewhat complicated and I did that a loooong time ago :( 

I just realized that the debugging method with Wireshark in parallel doesn't seem to work, at least not with my setup, because the socket seems to be exclusive. However, that depends a bit on the precise setup. 

This is how the output looks like on my Nexus 5 with the most recent version of InternalBlue:

```
internalblue$ python3 -m internalblue.cli
[!] pwnlib is not installed. Some features will not work.
[!] Opening a local Bluetooth socket failed. Not running on native Linux?
[*] No connected HCI device found
[*] Found multiple adb devices
[*] Connected to 0afb120602b358e7
[*] Chip identifier: 0x6109 (003.001.009)
[*] Using fw_0x6109.py
[*] Loaded firmware information for BCM4335C0.
[*] Try to enable debugging on H4 (warning if not supported)...
[*] Starting commandLoop for self.internalblue <internalblue.adbcore.ADBCore object at 0x7f410503f880>
   ____     __                    _____  __ 
  /  _/__  / /____ _______  ___ _/ / _ )/ /_ _____
 _/ // _ \/ __/ -_) __/ _ \/ _ `/ / _  / / // / -_)
/___/_//_/\__/\__/_/ /_//_/\_,_/_/____/_/\_,_/\__/

type <help -v> for usage information!
> log_level debug
[*] New log level: DEBUG
> hd 0x200400
[!] readMem: reading at 0x200400
[!] _sendThreadFunc: Send: 0108004dfc0500042000fb
[!] recvThreadFunc: received bt_snoop data 0000000900000009000000020000000000dcddef7d71b23b
[!] sendHciCommand.recvFilterFunction: got response
[!] recvThreadFunc: received bt_snoop data 0000010200000102000000030000000000dcddef7d71bd58
[!] sendHciCommand.recvFilterFunction: got response
[!] _sendThreadFunc: Send: 0108004dfc05fb04200005
[!] recvThreadFunc: received bt_snoop data 0000000900000009000000020000000000dcddef7d71d56f
[!] sendHciCommand.recvFilterFunction: got response
[!] recvThreadFunc: received bt_snoop data 0000000c0000000c000000030000000000dcddef7d71daac
[!] sendHciCommand.recvFilterFunction: got response
00200400: ff 1b dd 07  00 00 00 00  09 61 44 65  63 20 31 31   |····|····|·aDe|c 11|
00200410: 20 32 30 31  32 00 18 92  fc 00 3f 1f  00 00 00 00   | 201|2···|··?·|····|
00200420: 00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   |····|····|····|····|
00200430: 00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   |····|····|····|····|
00200440: 00 00 00 00  01 0a 03 00  40 04 60 00  98 00 00 00   |····|····|@·`·|····|
00200450: b1 25 0d 00  00 00 00 00  00 00 00 00  75 8a 00 00   |·%··|····|····|u···|
00200460: 8f 8a 00 00  5c 6f 21 00  9c 73 21 00  1c 74 21 00   |····|\o!·|·s!·|·t!·|
00200470: 70 04 20 00  70 04 20 00  03 00 0d 00  00 00 00 00   |p· ·|p· ·|····|····|
00200480: 00 00 00 00  53 8b 00 00  00 00 00 00  9f 8e 00 00   |····|S···|····|····|
00200490: 00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   |····|····|····|····|
002004a0: 00 00 00 00  00 00 00 00  10 01 a4 04  00 00 00 00   |····|····|····|····|
002004b0: 01 00 00 00  01 00 00 00  01 00 00 00  00 00 00 00   |····|····|····|····|
002004c0: 40 40 40 40  40 06 06 06  06 06 06 06  06 06 06 00   |@@@@|@···|····|····|
002004d0: 00 04 00 00  00 04 00 00  00 04 00 00  00 04 00 00   |····|····|····|····|
002004e0: 00 04 00 00  00 04 00 00  74 ff df 01  03 00 00 00   |····|····|t···|····|
002004f0: 00 00 02 04  17 10 00 01  00 00 00 00  3c 00 00 00   |····|····|····|<···|
> connect 13:37:ca:fe:ba:be
[!] _sendThreadFunc: Send: 01100005040dbebafeca371300000000000001
[!] recvThreadFunc: received bt_snoop data 0000001100000011000000020000000000dcddef7e69bc0c
[!] sendHciCommand.recvFilterFunction: got response
[!] recvThreadFunc: received bt_snoop data 0000000700000007000000030000000000dcddef7e69c911
[!] sendHciCommand.recvFilterFunction: got response
[*] [Connection Create initiated]
> [!] recvThreadFunc: received bt_snoop data 0000000e0000000e000000030000000000dcddef7eb7ef8e
[*] [Connect Complete: Handle=0xc  Address=1337cafebabe  status=Page Timeout]
```

Depending on what you want to do you can also work around via InternalBlue instead of fixing your Android `bluetooth.module.so` compiler errors.

## Workaround 1: Use the serial TTY interface within Android instead of modifying the driver.

For this, you can call InternalBlue with the `-s` parameter, which in turn calls the following code:
https://github.com/seemoo-lab/internalblue/blob/070f82844b9b211e4491ff10e9d18b9abcde2f9c/internalblue/adbcore.py#L347

Note that I did not test these commands on Android 6, only on 8-10. Thus, you might need to adapt them. But in general, you can always echo raw bytes into the Bluetooth serial interfaces. Forwarding this via `netcat` etc. is just much slower than changing the driver module.

You cannot enable the DIAG mode like this, because then you'd still get messages that are either ignored by Android or crash the driver. DIAG mode is the `0x7` thing that later on enables you to see the LMP traffic etc.

## Workaround 2: Disable DIAG

Assuming that your recompiled driver works (or the one you downloaded from here) except from diagnostics, you can also just remove this function:
https://github.com/seemoo-lab/internalblue/blob/070f82844b9b211e4491ff10e9d18b9abcde2f9c/internalblue/core.py#L2157

Hope that helps you on the way to fixing your issues :) And good luck with compiling Android ;)

Best,
Jiska

jurek4321, 2021-07-22:
Hi @jiska2342, I successfully build bluetooth.default.so for z3c, but i still have the same problem that i have with ready build bluetooth.default.so for nexus 5. When im trying to receive with lmp monitoring enabled im getting errors in logcat and bluetooth is restarting:

`05-29 03:20:47.488  2085  2712 W %s legacy transmit of command. Use transmit_command instead.: transmit_downward
05-29 03:20:50.775  2085  2706 W bt_hci_packet_fragmenter: reassemble_and_dispatch got continuation for unknown packet. Dropping it.
05-29 03:20:50.782  2085  2895 W bt_btm  : btm_acl_created hci_handle=12 link_role=1  transport=1
05-29 03:20:50.782  2085  2895 W bt_l2cap: L2CAP got conn_comp for unknown BD_ADDR
05-29 03:20:58.784  2085  2272 E bt_hci  : command_timed_out hci layer timeout waiting for response to a command. opcode: 0x41d
05-29 03:20:58.784  2085  2272 E %s restarting the bluetooth process.: command_timed_out
05-29 03:20:58.784  2085  2272 I %s      : ssr_cleanup
05-29 03:20:58.786  2085  2272 E bt_hci  : hci_cmd_timeout: SOC Status is reset
05-29 03:20:58.786  2085  2272 E bt_hci  :  
05-29 03:20:58.827   338   338 I brcm-uim: brcm-uim:After Polling to check POLLERR | POLLHUP erro = 1
05-29 03:20:58.827   338   338 I brcm-uim: brcm-uim:Breaking out from RE_POLL_TILL_POLL_ERR while loop with err=1
05-29 03:20:58.827   338   338 I brcm-uim: brcm-uim:value of install = 0
05-29 03:20:58.827   338   338 I brcm-uim: brcm-uim:value of dev_fd = 6
05-29 03:20:58.827   338   338 I brcm-uim: brcm-uim:snoop_enable = 0
05-29 03:20:58.828  2149  2149 D BluetoothPan: BluetoothPAN Proxy object disconnected
05-29 03:20:58.829  2149  2149 D PanProfile: Bluetooth service disconnected
`
Do you have any idea why its still not working and how i can fix it?

jiska2342, 2021-07-26:
Hi @jurek4321 ,
Android has minor versions. For example, Android 6.0.1 has multiple releases for the Nexus 5 (see https://source.android.com/setup/start/build-numbers). Most likely, you're using an incompatible pre-built `bluetooth.default.so`. And yes, it might work a bit if you're using the wrong version, just as your log shows, but fail later on.

jurek4321, 2021-07-26:
@jiska2342 , i know that it may not work, but as i wrote before i build bluetooth.default.so for z3c. I build it for lineage 14.1 and i'm using it with lineage 14.1. So it looks like patch for nexus 5 is not working with z3c. Is it possible that patch for nexus5 will not work with z3c? Did i need to change something it patch?

jiska2342, 2022-11-29:
Most likely, you will have to build `bluetooth.default.so` from scratch for your device and the correct LineageOS version. There are too many combinations of hardware and Android versions to provide pre-built Bluetooth drivers for all of them :(