seemoo-lab/internalblue #42: Adapting BLE PoC

RCayre, 2021-03-01:
Hi, 
i'm trying to adapt the BLE PoC for the CYW20735 evaluation board, but I'm a bit confused. I don't understand why multiple functions are hooked, _connTaskRxDone won't be enough ? What's the difference with the hook used for nexus 5 ? 

Thanks a lot for this great work !

jiska2342, 2021-03-02:
Hi,

the BLE PoC on the CYW20735 board was compiled with Nexmon so there's only some binary left of a version that worked for me ;) It's quite plausible that it contains too many hooks, since I usually set more hooks while developing a PoC than in the final version. Hooking `_connTaskRxDone` should definitely be sufficient for showing the currently active channel and channel map. If you need to modify the hook, it's probably easier to start with the Nexus 5 patch and replace the branches and locations of the referenced functions (`_connTaskRxDone`, `memcpy`, etc.). However, ideally compare how to send HCI packets with the PoC for the RNG, since the function signature / arguments and how to allocate an HCI buffer changed between the Nexus 5 and the CYW20735 board.

Best,
Jiska

jiska2342, 2021-03-02:
I just found the file that I think I was using with Nexmon (the `bluetooth-wip` branch) to create the initial PoC. Warning, contains some unnecessary code, weird comments, etc., but might help you with doing whatever you're working on :)

[patch.c.zip](https://github.com/seemoo-lab/internalblue/files/6070426/patch.c.zip)

RCayre, 2021-03-02:
Thanks a lot for this quick answer :) I'm trying to patch _scanTaskRxDone to send the received LL packet to the host using a custom HCI event.

RCayre, 2021-03-04:
Thanks to your nexmon patch, I managed to get the data i needed, thanks :) have you identified how the frequency / channel is configured in hardware ? I guess there is another memory mapped I/O allowing to configure it but i didn't find it.

jiska2342, 2021-03-04:
I'm not 100% sure, sorry. Malte Limmeroth reverse-engineered parts of that for his thesis "Bluetooth Low Energy Sniffing", supervised by Bastian Bloessl and me, which is only published in print in our university's library. I pasted the relevant parts for reference :)

The full process is rather complex, but you might have sufficient Bluetooth experience to figure it out ;)

[image]
[image]
[image]

RCayre, 2021-03-04:
omg, it's perfect. Thanks a LOT.
I would be **really** happy to read this thesis as soon as it will be available online !

RCayre, 2021-03-07:
I have another question, sorry ^^ Do you know other broadcom/cypress chips compatible with internalblue that supports Bluetooth 5, especially LE 2M Phy ? I'm mainly looking for a smartphone supporting this kind of features. I guess nexus 5 chip is too old to support LE 2M, right ?

jiska2342, 2021-06-17:
Hey, sorry for not answering the ticket, somehow got lost in my inbox :(

I think the Samsung Galaxy S10/S20 should support the LE 2M layer.

Did you manage to add the patches you needed etc.?

RCayre, 2021-06-17:
Hi, no problem, I finally used a Samsung Galaxy S20,  you are right, BCM4375 supports LE 2M!
I managed to implement all my patches, thanks. I'll probably release the code soon, I'll send you an email when it's public :)

jiska2342, 2021-06-17:
Okay, sounds great, so I'll close the ticket for now :)

closed 2021-06-17 by jiska2342