seemoo-lab/internalblue #44: Strange behaviour of CYW920819EVB-02

AntonFirc, 2021-04-01:
Hello, 

I noticed that when using the CYW920819EVB-02 dev board with internalblue on linux, the chip identifier has a strange value of `0x2305`, and the initialisation then falls back to using default `fw.py` file. 

I browsed trough the available files and identified that the `fw_0x220c.py` should contain the values for this board, thus I copied the file and renamed it to `fw_0x2305.py` to get matched with the board.  However I am not sure that this approach is correct as well as the values.

My questions thus is, why does the chip identifier differ?

Thank you very much!

jiska2342, 2021-04-01:
Looks like you got a chip with a more recent patch level :) Ideally, it should have the same ROM, so you can simply copy the file as you did. The most important definitions in the firmware file define the memory layout and functions in the ROM, which will not change with firmware updates.

On the evaluation boards, the patches, which are usually shipped as `.hcd` file, are contained in a separate flash memory on the board itself. Thus, if Cypress applied a newer patch level during the manufacturing process within that flash or if you installed a new version of the Cypress toolchain and applied those patches, the version number differs, even if you do not active load a `.hcd` file during chip initialization via Linux.

AntonFirc, 2021-04-02:
Ah, thanks for the explanation..I'm a little wiser again :)

Is there a way to verify that the ROM layout remained the same as in `fw_0x220c.py`? I'm currently experiencing some difficulties executing BIAS, so I'm wondering whether the different chip version might be the problem.

SDMJH, 2021-04-16:
> Ah, thanks for the explanation..I'm a little wiser again :)
> 
> Is there a way to verify that the ROM layout remained the same as in `fw_0x220c.py`? I'm currently experiencing some difficulties executing BIAS, so I'm wondering whether the different chip version might be the problem.

hello,
I also want to perform bias attack and i encountered the same problem as you, do you know how to solve it ？
thank you！

hnznhr, 2023-12-12:
hello,
I also want to perform bias attack and i encountered the same problem as you, do you know how to solve it ？
By the way, I use CYW920819EVB-01.