seemoo-lab/internalblue #46: Samsung galaxy S20 / dumpmem is failing

RCayre, 2021-04-14:
Hello, 
I'm trying to use internalblue to dump its firmware and patch a samsung galaxy s20 B4375B1 chip. However, the dumpmem command systematically fails at 80% with the following traceback :
`> dumpmem -f ~/samsunggalaxys20/firmware.bin
[*] No template found. Need to read ROM sections as well!
[ ] Initialize internal memory image: receiving data... 2031476 / 2523133 Bytes (80%)
[!] Received S10 Stack-Dump Event (contains 35 registers):
[!] pc: 0x0000b344   lr: 0x0000b3cb   sp: 0x00308000   r0: 0x00307f78   r1: 0x000000f9
r2: 0x00287b48   r3: 0x00000071   r4: 0x00308000   r5: 0x00287bd0   r6: 0x00000000

Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib64/python3.9/threading.py", line 950, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.9/threading.py", line 888, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/site-packages/internalblue/adbcore.py", line 266, in _recvThreadFunc
    callback(record)
  File "/usr/local/lib/python3.9/site-packages/internalblue/hci.py", line 948, in recvPacket
    self.handleS10StackDump(hcipkt)
  File "/usr/local/lib/python3.9/site-packages/internalblue/hci.py", line 1192, in handleS10StackDump
    self.finishStackDump()
  File "/usr/local/lib/python3.9/site-packages/internalblue/hci.py", line 970, in finishStackDump
    dump = flat(self.memdumps)  # flatten, as we have one entry per address chunk
TypeError: flat() missing 1 required positional argument: 'filler'
[!] _sendThreadFunc: No response from the firmware.
[!] sendHciCommand: waiting for response timed out!
[!] readMem: No response to readRAM HCI command! (read_addr=307f76, len=7ffff)
[!] _sendThreadFunc: No response from the firmware.
[!] sendHciCommand: waiting for response timed out!
[!] readMem: No response to readRAM HCI command! (read_addr=307f76, len=7ffff)
[!] _sendThreadFunc: No response from the firmware.
[!] sendHciCommand: waiting for response timed out!
[!] readMem: No response to readRAM HCI command! (read_addr=307f76, len=7ffff)
[!] _sendThreadFunc: No response from the firmware.
[!] sendHciCommand: waiting for response timed out!
[!] readMem: No response to readRAM HCI command! (read_addr=307f76, len=7ffff)
[!] readMem: failed!
EXCEPTION of type 'TypeError' occurred with message: 'cannot convert 'NoneType' object to bytes'
To enable full traceback, run the following command: 'set debug true'
`

jiska2342, 2021-04-14:
Looks to me like a crash in the `read_ram` function of the firmware. So, in the register state above you can see the PC which probably points to the `read_ram` function, and the arguments it has been passed, which include the address `0x00308000` to be read from that fails with the length argument that is probably `0x000000f9` (0xff minus what is needed for the HCI packet header).

Most likely this is because this memory is just not mapped, either permanently or just at the moment. The latter happens on some hardware peripherals. You can adjust the memory map in the `fw/...py` and remove that section. Most probably the part that you cannot read here is actually in the hardware peripheral area and not relevant for a ROM+RAM dump.

Moreover, on the S20/S10 chip series, echoing with a shell script into the serial device and then parsing the `btsnoop_hci.log` in Python to extract the RAM dump usually works faster, less delay than the Python+netcat stuff in InternalBlue :)

RCayre, 2021-04-15:
Thank you very much ! 
problem solved by commenting the last memory region in fw/fw_0x1111.py.

closed 2021-04-15 by RCayre