seemoo-lab/internalblue #48: Modify incoming LMP messages

drewbug, 2021-04-22:
Thank you for creating this extraordinary software and releasing it publicly.

Is it possible to modify incoming LMP messages before they are parsed by the chip? I'd particularly like to modify received `LMP_features_res` packets.

jiska2342, 2021-04-23:
Hi @drewbug,

thanks for correcting the typo :)

So, I think this should be possible, and most packet parsers in the firmware even have a pre hook to add handlers for certain types etc. Depends a bit on the parser and the firmware version, though.

Any particular firmware version on that you need this hook? I can try to give it a try somewhen this weekend :)

Best,
Jiska

drewbug, 2021-04-23:
Thank you! I'm using a Raspberry Pi Zero W so I think that's `BCM43430A1`.

jiska2342, 2021-04-23:
Hi @drewbug,

I just checked my pile of hardware and found a Raspberry Pi 3. I think it has the same chip as the Zero W but I'm not 100% sure. Could you please send me the first lines of the InternalBlue output where it says which firmware/chip it's using to let me confirm we have the same chip?

Best,
Jiska

drewbug, 2021-04-24:
I'm unfortunately about to board a transatlantic flight and my Pi is stowed away in my checked luggage. I'll get that information to you as soon as possible after landing. Thank you so much.

jiska2342, 2021-06-17:
Hi @drewbug  :) I assume this is still relevant? At least filtering incoming LMP can be useful for a couple of experiments. I have done some HCI filtering on the host side recently and it was super useful to confirm a bug in all major operating systems ;)

I'll probably implement LMP filtering for the WiSec 2021 tutorial, since it's the most useful feature request in the open tickets. If you could check again which chip you have that would help.

jiska2342, 2021-06-22:
I added an LMP filter example for the CYW20735 board. Since porting to other chips is always some work, I still need to know the precise chip you need the patch for. Or you can try to adapt it on your own :)

The patch for the CYW20735 board is available here:

https://github.com/seemoo-lab/internalblue/blob/master/examples/eval_cyw20735/LMP_Filter_PoC.py

closed 2022-11-29 by jiska2342