seemoo-lab/internalblue #49: Support for RaspberryPi 400 / BCM4345C3 (0x6606)

phantomblot-x, 2021-04-24:
Any chance that support for RaspberryPi 400 / BCM4345C3 (0x6606) will be added?

jiska2342, 2021-04-24:
Hi @phantomblot-x,

you should be fine by copying another `fw_...py` configuration file within the `fw` folder from a chip with a name close to BCM4345C3. Typically, the RAM/ROM layout doesn't change that much. If the chip crashes while dumping ROM/RAM, adjust the regions until it works.

Is there any specific feature you're looking for? For example, anything related to KNOB/BIAS requires reverse-engineering and diffing the firmware, which is quite a lot of effort (multiple days if you never did it before, but still a couple of hours depending on the chip's bugs if you know what you're doing). 

Best,
Jiska

phantomblot-x, 2021-04-24:
Hi @jiska2342 ,

Thanks for the quick response. My interest is in the LMP/LL packet injection, using it for fuzz-testing BT controllers from the air-interface and possibly for crafting some attacks.

I have tried to use BCM4345C0 (0x6119) but the chip crashes when attempting to inject a message. So, I guess some reverse engineering will be needed to make this work. I have no experience doing that for these chipsets. I was just hoping since all the other Raspberry Pi versions seem to be supported, that it might be added for the 400 version too.

jiska2342, 2021-04-25:
Hi,

LMP fuzzing should just work. There's a HCI command that can send LMP PDUs (0xFC58):
https://github.com/seemoo-lab/internalblue/blob/master/internalblue/hci.py#L340

...afair we even linked that to a `sendlmp` command, not sure if that made it through all the updates, but it definitely used to be a feature.

However, sending LMP via HCI is limited to a maximum of 17 bytes, since this is the maximum LMP PDU length. The firmware does some internal checks. If you want to bypass these, you need to write your own hook similar to the one we have for the Nexus 5 firmware. The same goes for sending LCP/LL (BLE) control packets, you need a custom patch similar to the one in the firmware you linked. Broadcom changed packet handling behavior from time to time, so the patch needs at least to be adapted to the changed addresses in ROM, but sometimes even more.

So, the simplest solution might be to only fuzz LMP and not LCP or get the Raspberry Pi that has support for it. Otherwise, you can try to dump the ROM and use Polypyus or BinDiff to find the handlers and modify them accordingly.

Best,
Jiska