seemoo-lab/internalblue #53: Internalblue Typerror at startup on Raspeberry pi 4b

delmarocks, 2021-06-24:
Does anyone encounter this error on raspeberry pi 4 b when running internalblue cli? I follow all the installation process in the documentation but still this error shows up. 

[image]

cchaine, 2021-06-24:
This issue is due to a change in the last version of `cmd2`. You can downgrade to `cmd2<=1.5.0` to solve this issue.

jiska2342, 2021-06-24:
Thanks @cchaine for pointing this out. I already bought a Raspberry Pi 4B these days and wanted to check if InternalBlue runs on it, so I might as well run into the same problem :)

delmarocks, 2021-06-24:
Thankyou @cchaine it did works by downgrading cmd2 to 1.5.0. 🙂

[image]

cchaine, 2021-06-24:
No worries. This problem has been fixed on master. You can close this issue.

jiska2342, 2021-06-25:
Checked everything with @cchaine 's pull request on a Raspberry Pi 4B and it just worked for me :) Tested on an up-to-date Raspberry Pi OS Lite.

However, I noted that reading the Patchram Enabled Bitfield does no longer work with the current firmware. The return status of HCI is `Status: Invalid HCI Command Parameters (0x12)` and thus, commands like `info patchram` just time out. That is definitely a firmware issue, but depending on what you want to do on that chip, you might need to downgrade the firmware patch file.

jiska2342, 2021-06-25:
Okay, quick fix to downgrade the firmware on the Raspberry Pi 4B:

```
cd /usr/lib/firmware/brcm/
cp BCM4345C0.hcd BCM4345C0_orig.hcd
wget https://github.com/RPi-Distro/bluez-firmware/blob/96eefffcccc725425fd83be5e0704a5c32b79e54/broadcom/BCM4345C0.hcd?raw=true
mv 'BCM4345C0.hcd?raw=true' BCM4345C0.hcd
```

Then reboot to load the old firmware. I will update the readme as well :)

closed 2021-06-25 by jiska2342