seemoo-lab/internalblue #54: Receives "socket.timeout: timed out"

delmarocks, 2021-06-29:
I'm currently facing an issue when connecting my rooted nexus 5 on raspberry pi 4 and running the internalblue I already push the patched bluetooth.default.so file via adb and still this error persist I can't figured out what's really the issue here. Thanks in advance.

[image]

jiska2342, 2021-07-07:
Hi,

socket timeout in this context could mean various things. Either something is wrong with your `adb` setup or something is wrong with the Bluetooth module on Android. You should be able to `adb shell` into your Android, run `su` and see open ports, including those by the Bluetooth module with `netstat`. InternalBlue tries to forward these ports and access them. In general, this should even work as a regular user on your Raspberry Pi, as long as the user can access `adb`.

jiska2342, 2021-07-07:
I just tried it with the most current version of InternalBlue while running on a Debian host. Output should look like this:

```
[*] No connected HCI device found
[*] Found multiple adb devices
[*] Connected to 0afb120602b358e7
[*] Chip identifier: 0x6109 (003.001.009)
[*] Using fw_0x6109.py
[*] Loaded firmware information for BCM4335C0.
[*] Try to enable debugging on H4 (warning if not supported)...
[*] Starting commandLoop for self.internalblue <internalblue.adbcore.ADBCore object at 0x7fe8e1780cd0>
   ____     __                    _____  __ 
  /  _/__  / /____ _______  ___ _/ / _ )/ /_ _____
 _/ // _ \/ __/ -_) __/ _ \/ _ `/ / _  / / // / -_)
/___/_//_/\__/\__/_/ /_//_/\_,_/_/____/_/\_,_/\__/

type <help -v> for usage information!
> 
```

Note that I had to turn off Bluetooth in the settings and then turn it on again, happens from time to time.

Ports should include the following:

```
hammerhead:/ # netstat -t                                                      
Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
tcp        0      0 localhost:8872          localhost:42082         ESTABLISHED
tcp        0      0 localhost:8873          localhost:58316         ESTABLISHED
...
```

Most likely, the pre-compiled Bluetooth module and your Android version don't match.

delmarocks, 2021-07-15:
@jiska2342 Thankyou for pointing out. I'll try to follow what you've said.

delmarocks, 2021-07-15:
@jiska2342 It did work. Thankyou😊😊😊[image]

closed 2021-07-26 by jiska2342