seemoo-lab/internalblue #66: Monitor LMP packets | Best hardware

warlock2k, 2022-11-29:
I plan to conduct braktooth attacks on the rooted android device and would like to monitor the LMP packets. Is this possible, if yes, which device would you suggest for this exercise?

jiska2342, 2022-11-29:
The best hardware for monitoring is probably the sending device, which should be an ESP32 for BrakTooth.

On an Android device, you probably won't find an ESP32. And if using InternalBlue with a Broadcom chip, when sending malicious payloads for the Baseband layer, some of the hardware configuration bits behave differently than on an ESP32, so you can't replicate exactly the same behavior. And even if the behavior is close enough, e.g. the LMP layer attacks, you would have to do a bit of reverse engineering. InternalBlue has a `sendlmp` option, but the Broadcom firmware applies further checks before sending the LMP packet, so you will have to write Assembly and binary patch the firmware to launch attacks.

Simply put: I wouldn't recommend using an Android phone for this. It's way too much effort, probably without achieving exactly the same results.

warlock2k, 2022-11-29:
Thank you for your response, and taking the time.

I have a ESP32 WROVER kit and I'm already monitoring the LMP data from the sending device. All I want to do now is know if there is a way to capture the same packets over the air bluetooth packets between the sending and attacked device using internal blue as a sniffer.

warlock2k, 2022-11-30:
@jiska2342 What I'm hoping to do is to collect some normal Bluetooth traffic on the attacked device along with the attack data. Do you think this is feasible?

jiska2342, 2022-12-01:
Hi :)

InternalBlue does not record packets over-the-air, you would need an Ubertooth for that. However, Ubertooth also doesn't seem to be great for BT Classic, and I think there's still no good non-commercial sniffing solution.

If you want to capture LMP on the receiving device, you can use Broadcom's Diagnostic feature. It's also described in [this paper](https://arxiv.org/pdf/1905.00634.pdf). In short, what Diagnostics does: It adds a new type 0x7 in addition to the types 0x1/0x2/0x3/0x4 resembling ACL/SCO/HCI over UART. After enabling, it will send Diagnostics with the prefix type 0x7 over UART. These Diagnostics include LMP packets as they are received. However, there is no parsing/validation. You'll get 63 bytes of LMP buffer for every packet, despite the maximum specified length for LMP packets being 17 bytes. Also, if BrakTooth corrupts something inside the Modem (aka Baseband layer) or if a corruption happens before copying the malicious LMP packet to the diagnostic buffer, you might not see it. But at least you can verify that everything *before* arrived and was processed correctly :)

While Linux and even some macOS versions have built-in Diagnostics mode for Broadcom chips, this is not the case for Android. Once you enable diagnostics, AFAIR, the Bluetooth driver is crashing since somewhen around Android 10. However, I haven't tired newer releases, and maybe they added support again. It might be that you will have to modify drivers.

Regarding devices, which was your initial question: Samsung phones with an Exynos chipset built in Europe have Broadcom Wi-Fi/Bluetooth, such as the Samsung Galaxy S20/21/22. Google Pixel and Samsung US models have Qualcomm chips, and the cheaper Samsung series have Mediatek as far as I know, so none of them are an option.

Hope that helps :) Best case, you'll only have to enable Diagnostics and patch a bit in a Samsung Android driver, if you really only want to receive LMP.

warlock2k, 2022-12-02:
Perfect, this makes a lot of sense. I will explore the diagnostics feature as described in the paper. I assume that I can also capture LMP packets by setting up a Nexus 5 device with internal blue (as described in the original thesis) and attacking it from the ESP32 WROVER kit to capture the LMP packets?

Thanks a lot for taking the time, this is useful for my research.

warlock2k, 2022-12-03:
@jiska2342 Thanks a lot, I was able to accomplish all that I needed. You're a star ⭐

closed 2022-12-03 by warlock2k