seemoo-lab/internalblue #69: Regarding Frida on Iphone 8 for key exchange!

abbiy, 2023-03-22:
Hello i am trying to perform key change using frida on iphone 8. But my ios version is 13.5.1 and the code given supports 14.2,14.4 iphone8 . Can anyone tell me how to know the base addresses? 
var OI_HCIIfc_DataReceived = base.add(0xee5a4);  // iOS 14.1, iPhone 12
// var OI_HCIIfc_DataReceived = base.add(0xed9f8);  // iOS 14.8, iPhone 8
// var OI_HCIIfc_DataReceived = base.add(0xed0b8);  // iOS 14.4, iPhone 8
How exactly can i calculate the these base address values of my ios 13.5 , iphone 8.

jiska2342, 2023-05-04:
Ideally, download the bluetoothd of one of the versions with known addresses. You can find these on https://appledb.dev, unzip the IPSW, and then mount the largest .dmg file. Within that, you can find the bluetoothd binary.

For diffing, you can use Ghidra's versioning tool or attempt to manually do so. Many functions in bluetoothd still print meaningful strings and typically functions are easy to find, unless Apple did too much refactoring between the versions.

abbiy, 2023-06-26:
Hi jiska thanks for your reply. I tried the poc using the same steps and setup of keychange but whenever i change the LTK it doesn't get connected and again target device asked for pairing process. Can you tell me is there any way possible to achieve the poc?